balena-os/meta-balena #1575: Add support to allow the user to configure usb device scan during u-boot

ZubairLK, 2019-07-02:
We disabled usb scanning in u-boot to speed up the boot by ~2 seconds.. balena-os/meta-balena#1432

Some people would like to boot from usb. See https://github.com/balena-os/balena-raspberrypi/issues/341 for more discussion

I guess we can add a flag in `resinOS_uEnv.txt` (or somewhere rollback proof) and if someone wants usb scanning, they can manually enable it via there which will be checked via `env_resin.h`

komikoni, 2019-07-07:
How do I set resinous uEnv.txt?
Do you have a tutorial?

There is no detailed information in the listed Link.

floion, 2019-07-08:
@komikoni right now this isn't implemented so there's no indications to be given to achieve this

damianpowell, 2019-09-02:
Hi, folks. Is there any progress on this?

ZubairLK, 2019-09-19:
Not yet I'm afraid.

ZubairLK, 2019-10-09:
I was looking at booting balenaOS from usb and would like some more information..

There are 2 ways of booting from usb.
a) Completely boot from usb including bootloader
b) Having the bootloader on mmc/eMMC and then booting rootfs from USB.

e.g. Option a. Completely boot from USB for Raspberry Pi would require setting the OTP for usb boot. https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/msd.md

Option b. Having bootloader on mmc means that there is still an sd card plugged in the raspberry pi. But then you wanted to boot the rest of the rootfs from USB via some way. (how did you go about doing that previously if you used this method?)

cc @komikoni @damianpowell @nunux13 @markhaehnel @shaunmulligan @imrehg

ZubairLK, 2019-10-09:
follow up question as well. Are people looking for the raspberry pi family of devices only for usb boot?

ZubairLK, 2019-10-09:
lol. I forgot to ask the question in my earlier comment.
Which option were you guys using/trying before on v2.29?

lurch, 2019-11-16:
> Option a. Completely boot from USB for Raspberry Pi would require setting the OTP for usb boot. 

That's needed for the RaspberryPi 3B, but that OTP bit is enabled by default on RPi 3B+ boards.

ZubairLK, 2019-11-18:
> That's needed for the RaspberryPi 3B, but that OTP bit is enabled by default on RPi 3B+ boards.

@lurch thank-you for pointing that out! Back to the drawing board for a bit with this info..

ZubairLK, 2019-11-18:
To add some context to the above comment. 
We had discussed this internally and the plan of action was to use that otp bit as a marker in uboot to scan usb devices or not. (would like to keep usb scanning disabled by default due to security/performance concerns)

lurch, 2019-11-18:
@ZubairLK What if you only moved on to the USB-scanning phase if the SD card-slot was empty (or the eMMC has no MBR in the case of the CM boards) ?
If there's no SD card present, I guess that implies the bootloader must have been loaded from USB? :man_shrugging:

floion, 2019-11-18:
> > Option a. Completely boot from USB for Raspberry Pi would require setting the OTP for usb boot.
> 
> That's needed for the RaspberryPi 3B, but that OTP bit is enabled by default on RPi 3B+ boards.

Can you point to some info about this?

damianpowell, 2019-11-18:
@ZubairLK Apologies for not responding earlier. For my purposes, I'm looking to completely boot from an SSD. I'm doing this more for reliability than for speed, although the extra speed is useful. To further complicate things, I'd like to do this on a Raspberry Pi 4 which currently doesn't boot from USB at all! The lack of USB boot support on the RPi 4 is the reason I've not been paying attention lately, but is on the cards shotly.

My ultimate goal is to boot an RPi 4 directly from an SSD (or USB HDD) without any reliance upon an SD card but while we wait for the RPi 4, I am happy to proceed with the RPi 3 B+. I guess this means I want **option a**. However, I would not be opposed to option b, if we can be 100% certain that the SD card is operating in read only mode.

ZubairLK, 2019-11-18:
@damianpowell Thanks for the reply.

Regarding pi4. the USB is via pci-express. u-boot doesn't really have support for pci-express. And I'm guessing that is the barrier the pi foundation is facing as well.

But yea a small sd card that has only the necessary bootloader. rest OS on ssd should be doable. Although. this is an interesting use-case that complicates some of our Host OS update work.

lurch, 2019-11-18:
> > > Option a. Completely boot from USB for Raspberry Pi would require setting the OTP for usb boot.
> > 
> > 
> > That's needed for the RaspberryPi 3B, but that OTP bit is enabled by default on RPi 3B+ boards.
> 
> Can you point to some info about this?

https://github.com/raspberrypi/documentation/blob/master/hardware/raspberrypi/bootmodes/msd.md

https://www.raspberrypi.org/blog/raspberry-pi-3-model-bplus-sale-now-35/

floion, 2019-11-19:
> > > > Option a. Completely boot from USB for Raspberry Pi would require setting the OTP for usb boot.
> > > 
> > > 
> > > That's needed for the RaspberryPi 3B, but that OTP bit is enabled by default on RPi 3B+ boards.
> > 
> > 
> > Can you point to some info about this?
> 
> https://github.com/raspberrypi/documentation/blob/master/hardware/raspberrypi/bootmodes/msd.md
> (and [raspberrypi/documentation#1334](https://github.com/raspberrypi/documentation/pull/1334) and [raspberrypi/documentation#1250](https://github.com/raspberrypi/documentation/pull/1250) )
> 
> https://www.raspberrypi.org/blog/raspberry-pi-3-model-bplus-sale-now-35/

Thanks Andrew for the resources

nunux13, 2019-11-26:
Hello,

I'm using option a) full USB Boot without SDCard.
When building / testing a lot of OS images variations for validation,  I mean: burning it on mac, then swap, and also when making quick prototypes, you go faster with USB Keys. Sometime you need tools just to swap the SD ...

I think you should still consider to give back easy enablement for USB Boot.

Best,

ZubairLK, 2019-11-27:
Ok. We had another discussion during the arch call

The approach now is as follows
- The general strategy is: If we have information from bootrom/first stage bootloader that uboot/grub is running from usb, or there is some way of indicating a privileged action (i.e. press of a button for BBB, selecting usb boot in NUC bios etc.), then balenaOS should be able to boot from those USB devices. (note 'should be able to'. might not act this way atm)

- If we don't have that information from first stage bootloader/bootrom, we should try to default to the existing device bootrom/bios behaviour. e.g. for the rpi family the bootrom does the following, try to boot from sd card, if unable to, then try to boot from usb. And that is the behaviour we should emulate. i.e. If no mmc card detected (or I guess no `kernelX.img` is detected), then we should try usb devices. When implemented, the pi should be able to boot from usb.

cc @floion @acostach who were looking at the pi usb boot support

lurch, 2019-12-02:
> (or I guess no `kernelX.img` is detected)

IIRC `bootcode.bin` is actually the first file that the Raspberry Pi devices try to load.
https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/bootflow.md
https://www.raspberrypi.org/documentation/configuration/boot_folder.md

(with the exception of the Pi4)