balena-os/meta-balena #1579: v2.38.2 breaks Xorg

willswire, 2019-07-04:
since updating to v2.38.2, Xorg is unable to work properly. full output of /var/log/Xorg.0.log below:

`bash: /var/log/Xorg.0.log: Permission denied
root@e96eb07:/usr/src/app# cat /var/log/Xorg.0.log
[  2074.395] 
X.Org X Server 1.20.4
X Protocol Version 11, Revision 0
[  2074.395] Build Operating System: Linux 4.9.0-8-amd64 x86_64 Debian
[  2074.395] Current Operating System: Linux e96eb07 5.0.3-yocto-standard #1 SMP PREEMPT Thu Jun 27 19:24:06 UTC 2019 x86_64
[  2074.395] Kernel command line: BOOT_IMAGE=/boot/bzImage root=LABEL=resin-rootB rootwait quiet loglevel=0 splash udev.log-priority=3 vt.global_cursor_default=0 intel_idle.max_cstate=1
[  2074.395] Build Date: 05 March 2019  08:11:12PM
[  2074.395] xorg-server 2:1.20.4-1 (https://www.debian.org/support) 
[  2074.395] Current version of pixman: 0.36.0
[  2074.395]    Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
[  2074.395] Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[  2074.396] (==) Log file: "/var/log/Xorg.0.log", Time: Thu Jul  4 12:28:28 2019
[  2074.397] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[  2074.397] (==) No Layout section.  Using the first Screen section.
[  2074.397] (==) No screen section available. Using defaults.
[  2074.397] (**) |-->Screen "Default Screen Section" (0)
[  2074.397] (**) |   |-->Monitor "<default monitor>"
[  2074.399] (==) No monitor specified for screen "Default Screen Section".
        Using a default monitor configuration.
[  2074.399] (==) Automatically adding devices
[  2074.399] (==) Automatically enabling devices
[  2074.399] (==) Automatically adding GPU devices
[  2074.399] (==) Max clients allowed: 256, resource mask: 0x1fffff
[  2074.399] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[  2074.399]    Entry deleted from font path.
[  2074.400] (==) FontPath set to:
        /usr/share/fonts/X11/misc,
        /usr/share/fonts/X11/100dpi/:unscaled,
        /usr/share/fonts/X11/75dpi/:unscaled,
        /usr/share/fonts/X11/Type1,
        /usr/share/fonts/X11/100dpi,
        /usr/share/fonts/X11/75dpi,
        built-ins
[  2074.400] (==) ModulePath set to "/usr/lib/xorg/modules"
[  2074.400] (II) The server relies on udev to provide the list of input devices.
        If no devices become available, reconfigure udev or disable AutoAddDevices.
[  2074.400] (II) Loader magic: 0x559b1e670e20
[  2074.400] (II) Module ABI versions:
[  2074.400]    X.Org ANSI C Emulation: 0.4
[  2074.400]    X.Org Video Driver: 24.0
[  2074.400]    X.Org XInput driver : 24.1
[  2074.400]    X.Org Server Extension : 10.0
[  2074.400] (EE) dbus-core: error connecting to system bus: org.freedesktop.DBus.Error.FileNotFound (Failed to connect to socket /var/run/dbus/system_bus_socket: No such file or directory)
[  2074.400] (--) using VT number 2

[  2074.401] (II) systemd-logind: logind integration requires -keeptty and -keeptty was not provided, disabling logind integration
[  2074.405] (--) PCI:*(0@0:2:0) 8086:22b0:8086:7270 rev 54, Mem @ 0x90000000/16777216, 0x80000000/268435456, I/O @ 0x0000f000/64, BIOS @ 0x????????/131072
[  2074.405] (II) LoadModule: "glx"
[  2074.406] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[  2074.409] (II) Module glx: vendor="X.Org Foundation"
[  2074.409]    compiled for 1.20.4, module version = 1.0.0
[  2074.410]    ABI class: X.Org Server Extension, version 10.0
[  2074.410] (==) Matched modesetting as autoconfigured driver 0
[  2074.410] (==) Matched fbdev as autoconfigured driver 1
[  2074.410] (==) Matched vesa as autoconfigured driver 2
[  2074.410] (==) Assigned the driver to the xf86ConfigLayout
[  2074.410] (II) LoadModule: "modesetting"
[  2074.411] (II) Loading /usr/lib/xorg/modules/drivers/modesetting_drv.so
[  2074.411] (II) Module modesetting: vendor="X.Org Foundation"
[  2074.411]    compiled for 1.20.4, module version = 1.20.4
[  2074.411]    Module class: X.Org Video Driver
[  2074.411]    ABI class: X.Org Video Driver, version 24.0
[  2074.411] (II) LoadModule: "fbdev"
[  2074.412] (WW) Warning, couldn't open module fbdev
[  2074.412] (EE) Failed to load module "fbdev" (module does not exist, 0)
[  2074.412] (II) LoadModule: "vesa"
[  2074.413] (WW) Warning, couldn't open module vesa
[  2074.413] (EE) Failed to load module "vesa" (module does not exist, 0)
[  2074.413] (II) modesetting: Driver for Modesetting Kernel Drivers: kms
[  2074.426] (EE) open /dev/dri/card0: No such file or directory
[  2074.426] (WW) Falling back to old probe method for modesetting
[  2074.426] (EE) open /dev/dri/card0: No such file or directory
[  2074.426] (EE) Screen 0 deleted because of no matching config section.
[  2074.426] (II) UnloadModule: "modesetting"
[  2074.426] (EE) Device(s) detected, but none match those in the config file.
[  2074.426] (EE) 
Fatal server error:
[  2074.426] (EE) no screens found(EE) 
[  2074.426] (EE) 
Please consult the The X.Org Foundation support 
         at http://wiki.x.org
 for help. 
[  2074.426] (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
[  2074.427] (EE) 
[  2074.435] (EE) Server terminated with error (1). Closing log file.`

floion, 2019-07-05:
Hi, can you try this project please? https://github.com/balena-io-playground/x11-window-manager

willswire, 2019-07-05:
Trying a different project yields the same issue.  Rolling back to v2.38.0+rev1 fixes the issue.  Having to manually refresh all of these devices is a pain - any way we can rollback the update thru the Host OS?

floion, 2019-07-05:
There isn't the option to downgrade unfortunately. @imrehg is this planned?

@willswire so the same issues persists with `v2.38.1+rev1` or `v2.38.1+rev2` ?

willswire, 2019-07-05:
2.38.1+rev1 and 2.38.1+rev2 do not seem to cause any issues.

floion, 2019-07-05:
On what board is this? What balena OS are you running?

floion, 2019-07-05:
So is your board running the Intel NUC image?
And can you confirm that `v2.38.2+rev1` works and `v2.38.2+rev2` doesn't work?

willswire, 2019-07-05:
We are deploying to a Intel NUC application to both some [Mini PC Hardware](https://www.amazon.com/Z83-F-PC-X5-Z8350-Graphics-1000Mbps/dp/B07BF6K7RH) and some [Dell Optiplex 7010 Hardware](https://www.newegg.com/dell-optiplex-7010-business-desktops-workstations/p/N82E16883995016?Description=dell%20optiplex%207010%20desktop&cm_re=dell_optiplex_7010_desktop-_-83-995-016-_-Product).  I should mention that it appears `v2.38.2+rev1` works fine on the Dell hardware -- the issues we are having with Xorg only exhibit on `v2.38.2+rev1` on the Mini PC hardware.

willswire, 2019-07-05:
`v2.38.2+rev2` is not available within the balenaCloud control panel for the Intel NUC application.

floion, 2019-07-05:
Ok, thanks for the info.
So `2.38.1+rev2` works on the Mini PC and `2.38.2+rev1` does not work on the Mini PC? Trying to find the exact version that broke for your hardware

floion, 2019-07-05:
`v2.38.2+rev2` is an internal release we have in our staging environment, I thought you may be using that one

willswire, 2019-07-05:
> Ok, thanks for the info.
> So `2.38.1+rev2` works on the Mini PC and `2.38.2+rev1` does not work on the Mini PC? Trying to find the exact version that broke for your hardware

Correct!

> `v2.38.2+rev2` is an internal release we have in our staging environment, I thought you may be using that one

Is there any way we can test this update?

floion, 2019-07-05:
The update does not contain changes that might impact in your issue. Looking to see if there was any change between those versions that could have an impact

floion, 2019-07-05:
So in production we released `2.38.0+rev1` and `2.38.2+rev1` and between those there was a kernel bump to kernel 5.0.3. While we test on some hardware we cannot test on all existing generic x86-64 boards out there. And I suspect the Mini PC has a bug that prevents kernel 5.0.3 running Xorg for some reason

ZubairLK, 2020-01-27:
Closing this issue for cleanup. Please reopen if its still valid and with newer OS versions. Seems device specific.

closed 2020-01-27 by ZubairLK