balena-os/meta-balena #1604: Balena OS 2.38.0 rev1 Could not open moddep file

savnik, 2019-07-31:
When running Balena OS 2.38.0 rev1 we get an error locally on the device host when plugging in an external screen:
```
systemd-udevd[52]: could not open moddep file '/lib/modules/4.14.98/modules.dep.bin'
systemd-udevd[52]: could not open moddep file '/lib/modules/4.14.98/modules.dep.bin'
systemd-udevd[32]: could not open moddep file '/lib/modules/4.14.98/modules.dep.bin'
systemd-udevd[32]: could not open moddep file '/lib/modules/4.14.98/modules.dep.bin'
```
We believe this relates to some network issues we experience where the wireless connection dies after some time and is renaming eth0 in a loop. 
Running debmod gives following result:

```
root@0a55afd:~# depmod
depmod: ERROR: openat(/lib/modules/4.14.98, modules.dep.tmp, 1101, 644): Read-only file system
depmod: ERROR: openat(/lib/modules/4.14.98, modules.dep.bin.tmp, 1101, 644): Read-only file system
depmod: ERROR: openat(/lib/modules/4.14.98, modules.alias.tmp, 1101, 644): Read-only file system
depmod: ERROR: openat(/lib/modules/4.14.98, modules.alias.bin.tmp, 1101, 644): Read-only file system
depmod: ERROR: openat(/lib/modules/4.14.98, modules.softdep.tmp, 1101, 644): Read-only file system
depmod: ERROR: openat(/lib/modules/4.14.98, modules.symbols.tmp, 1101, 644): Read-only file system
depmod: ERROR: openat(/lib/modules/4.14.98, modules.symbols.bin.tmp, 1101, 644): Read-only file system
depmod: ERROR: openat(/lib/modules/4.14.98, modules.builtin.bin.tmp, 1101, 644): Read-only file system
depmod: ERROR: openat(/lib/modules/4.14.98, modules.devname.tmp, 1101, 644): Read-only file system
```
Looks like some modules is missing or the reference is wrong.

ZubairLK, 2019-08-02:
@savnik Which device is this?
Did you hup to this OS version or clean flash?

agherzan, 2019-08-02:
Also, what is the reason why you need depmod at runtime?

ZubairLK, 2020-01-27:
Closing stale issues for cleanup. Please reopen with more detail if you are still facing this.

closed 2020-01-27 by ZubairLK