balena-os/meta-balena #1619: Investigate /var/run warnings

ZubairLK, 2019-08-21:
Saw these in dmesg. Investigate..
```
systemd[1]: /lib/systemd/system/openvpn.service:11: PIDFile= references path below legacy directory /var/run/, updating /var/run/openvpn/openvpn.pid → /run/openvpn/openvpn.pid; please update the unit file accordingly.
systemd[1]: /lib/systemd/system/chronyd.service:9: PIDFile= references path below legacy directory /var/run/, updating /var/run/chrony/chronyd.pid → /run/chrony/chronyd.pid; please update the unit file accordingly.
```

imrehg, 2019-08-28:
Checkign 2.41.0+rev3 Raspberry Pi 3, I've seen a handful more (dev image I've checked)

```
root@8512092:~# journalctl -a --no-pager |grep "legacy directory"
Aug 08 09:59:55 localhost systemd[1]: /lib/systemd/system/openvpn.service:11: PIDFile= references path below legacy directory /var/run/, updating /var/run/openvpn/openvpn.pid → /run/openvpn/openvpn.pid; please update the unit file accordingly.
Aug 08 09:59:56 localhost systemd-tmpfiles[503]: [/etc/tmpfiles.d/balena-tmpfiles.conf:1] Line references path below legacy directory /var/run/, updating /var/run/docker.pid → /run/docker.pid; please update the tmpfiles.d/ drop-in file accordingly.
Aug 08 09:59:56 localhost systemd-tmpfiles[503]: [/etc/tmpfiles.d/balena-tmpfiles.conf:2] Line references path below legacy directory /var/run/, updating /var/run/balena.pid → /run/balena.pid; please update the tmpfiles.d/ drop-in file accordingly.
Aug 08 09:59:56 localhost systemd-tmpfiles[503]: [/etc/tmpfiles.d/nm-tmpfiles.conf:1] Line references path below legacy directory /var/run/, updating /var/run/resolvconf/interface → /run/resolvconf/interface; please update the tmpfiles.d/ drop-in file accordingly.
Aug 08 09:59:56 localhost systemd-tmpfiles[503]: [/etc/tmpfiles.d/nm-tmpfiles.conf:2] Line references path below legacy directory /var/run/, updating /var/run/resolvconf/interface/NetworkManager → /run/resolvconf/interface/NetworkManager; please update the tmpfiles.d/ drop-in file accordingly.
Aug 08 09:59:56 localhost systemd-tmpfiles[503]: [/etc/tmpfiles.d/openvpn.conf:1] Line references path below legacy directory /var/run/, updating /var/run/openvpn → /run/openvpn; please update the tmpfiles.d/ drop-in file accordingly.
Aug 08 10:01:27 localhost systemd-tmpfiles[1423]: [/etc/tmpfiles.d/balena-tmpfiles.conf:1] Line references path below legacy directory /var/run/, updating /var/run/docker.pid → /run/docker.pid; please update the tmpfiles.d/ drop-in file accordingly.
Aug 08 10:01:27 localhost systemd-tmpfiles[1423]: [/etc/tmpfiles.d/balena-tmpfiles.conf:2] Line references path below legacy directory /var/run/, updating /var/run/balena.pid → /run/balena.pid; please update the tmpfiles.d/ drop-in file accordingly.
Aug 08 10:01:27 localhost systemd-tmpfiles[1423]: [/etc/tmpfiles.d/nm-tmpfiles.conf:1] Line references path below legacy directory /var/run/, updating /var/run/resolvconf/interface → /run/resolvconf/interface; please update the tmpfiles.d/ drop-in file accordingly.
Aug 08 10:01:27 localhost systemd-tmpfiles[1423]: [/etc/tmpfiles.d/nm-tmpfiles.conf:2] Line references path below legacy directory /var/run/, updating /var/run/resolvconf/interface/NetworkManager → /run/resolvconf/interface/NetworkManager; please update the tmpfiles.d/ drop-in file accordingly.
Aug 08 10:01:27 localhost systemd-tmpfiles[1423]: [/etc/tmpfiles.d/openvpn.conf:1] Line references path below legacy directory /var/run/, updating /var/run/openvpn → /run/openvpn; please update the tmpfiles.d/ drop-in file accordingly.
```

ZubairLK, 2019-09-06:
There are still some(chrony/openvpn) that should be removed via package bumps in future Yocto versions.

These warnings should be innocuous though.

PR #1644 (2019-09-03): Fix systemd tmpfile warnings
Fixes #1619

---
### Contributor checklist

- [x] Changes have been tested
- [x] `Change-type` present on at least one commit
- [x] `Signed-off-by` is present
- [x] The PR complies with the [Open Embedded Commit Patch Message Guidelines](http://www.openembedded.org/wiki/Commit_Patch_Message_Guidelines)

### Reviewer Guidelines
- When submitting a review, please pick:
  - '*Approve*' if this change would be acceptable in the codebase (even if there are minor or cosmetic tweaks that could be improved).
  - '*Request Changes*' if this change would not be acceptable in [... cut]
closed 2019-09-06 by ZubairLK