balena-os/meta-balena #1621: Rpi Zero with with doesn't boot with the PiSupply Lora Hat

krasi-georgiev, 2019-08-21:
I have the Pi Zero W with the PiSupply Hat
https://uk.pi-supply.com/products/iot-lora-node-phat-for-raspberry-pi?_pos=10&_sid=e643e7799&_ss=r

Flashed 2.32.0+rev1-dev-v9 with the Dev image and wifi enabled.

the led starts blinking and than after few seconds just stays on all the time and that is it.

Don't have an HDMI monitor to check what is on the output.

removing "console=tty1 console=serial0,115200" fixes the problem.

@ryanteck  might give some clue on what could be the problem.

ryanteck, 2019-08-22:
Was this with the GPS module also plugged in?

krasi-georgiev, 2019-08-22:
you mean the Usb GPS module that I am also using?

No I had it disconnected, but also tried with the GPS connected and booted ok.

agherzan, 2019-08-22:
@krasi-georgiev can you attach a serial console and check the logs?

ryanteck, 2019-08-22:
After some looking in it was actually the pHAT and why I got confused.

The pHAT is UART based so will need the line changing for it to be usable. Not sure why it would have halted boot of the Zero though.

krasi-georgiev, 2019-08-23:
> @krasi-georgiev can you attach a serial console and check the logs?

not sure how would I do that?

ZubairLK, 2019-08-23:
@krasi-georgiev In the dev image, we use the uart for a serial console to send messages from the kernel and have a login shell as well.. I'm guessing the HAT also wants to use the same UART. This will clash. hence when you remove the console args from kernel cmdline (which disables the shell and frees up the uart), things start working.

It is a bit hard to exactly see what is halting boot with the console.
Perhaps the response sent back by the phat is received by the login shell and understood in a funny way.

The way forward would be disable the uart as you've done and proceed with your work. This shouldn't affect the production images as they don't send tty output on the uart.

krasi-georgiev, 2019-08-23:
Nice, thanks for the explanation!

closed 2019-08-23 by krasi-georgiev