balena-os/meta-balena #1626: Can't read from the serial port /dev/ttyAMA0

krasi-georgiev, 2019-08-23:
I have this simple tool
https://github.com/arribada/smartLoraTracker/blob/master/tools/GPSSender/main.go

It uses the Lora Phat rak811 module and when I try to run a docker container directly on the host(with `balena ssh 192.168.0.100`) it errors out with:
`write /dev/ttyAMA0: input/output error`

Here is the current Balena OS setup:

Rpi Zero W
```
uname -a
Linux e2f2e72 4.14.98 #1 Tue Apr 30 08:12:37 UTC 2019 armv6l armv6l armv6l GNU/Linux

cat /mnt/boot/cmdline.txt 
dwc_otg.lpm_enable=0 rootfstype=ext4 rootwait 

cat /mnt/boot/config.txt 
dtoverlay=pi3-miniuart-bt
enable_uart=1
disable_splash=1
gpu_mem=16

```
And running the docker image.
```
balena run -it --rm --privileged --name gpssender arribada/gpssender --dev_eui=$DEV_EUI --app_key=$APP_KEY

15:22:27 main.go:146: lora module reset resp: 
15:22:27 main.go:56: failed to create lora connection err:set lora mod: failed to write command "mode=0" with: write /dev/ttyAMA0: input/output error
```

the same setup works absolutely fine on Hypriot

ZubairLK, 2019-08-28:
@krasi-georgiev This is strange.. Your configuration seems sound to me. Are you using a `development` image or a `production` image? If it is a development image, maybe getty is running on that port and interfering..

From the host OS, can you `echo > /dev/ttyAMA0` ?
And can you run a standard debian container with the privileged flag and try the same echo above from inside the container?

krasi-georgiev, 2019-08-28:
yes development image, tested the production one and it works fine there.

I will try the command a bit latter

ZubairLK, 2019-08-29:
@krasi-georgiev so the `development` image is also running a getty login shell on the console.
It is probably that conflicting with your usage of serial uart?

krasi-georgiev, 2019-08-30:
probbably, I will try to disable and will let you know.

krasi-georgiev, 2019-09-29:
closing now as I am running the production image now, but will report back if/when I have to use the dev image.

closed 2019-09-29 by krasi-georgiev