balena-os/meta-balena #1627: kernel-module-build example fails for rpi3

floion, 2019-08-26:
[main]       CC      /usr/src/app/example_module_raspberrypi3_2.41.0+rev3.prod/hello.mod.o
[main]       LD [M]  /usr/src/app/example_module_raspberrypi3_2.41.0+rev3.prod/hello.ko
[main]     ld: cannot open linker script file ./arch/arm/kernel/module.lds: No such file or directory
[main]     
[main]     make[1]: *** [scripts/Makefile.modpost:129: /usr/src/app/example_module_raspberrypi3_2.41.0+rev3.prod/hello.ko] Error 1
[main]     
[main]     make: *** [Makefile:1523: modules] Error 2

This error is present on both the `master` branch or the `thud_or_newer` branch

ZubairLK, 2019-08-27:
strange. i checked the pi3. will investigate

ZubairLK, 2019-08-29:
https://github.com/balena-os/meta-balena/pull/1605 should fix it
specifically https://github.com/balena-os/meta-balena/pull/1605/commits/45ba60e6a210480444c487b6e6f496961a4b33fa

ZubairLK, 2019-09-16:
Should be fixed by https://github.com/balena-os/meta-balena/pull/1605

#1632 needs a bit of work to see the impact of released headers.

floion, 2019-09-17:
The kernel module build example needs to be updated also, currently it fails

ZubairLK, 2019-09-17:
@floion should be fine using https://github.com/balena-io-projects/kernel-module-build/pull/28

closed 2019-09-24 by ZubairLK