balena-os/meta-balena #1630: Namespace the hello-world health-check image

ZubairLK, 2019-08-27:
An unknown hello-world image can be confusing to people. Forgot to namespace the container with a `balena_` to clarify..

Update and namespace the image.

pdcastro, 2019-08-27:
Adding to @dt-rush's voice in [flowdock](https://www.flowdock.com/app/rulemotion/user_happiness/threads/bn1Ar9a9xHM7vBhCQ46B9tlnhSM), I think it would be good to not only prefix the hello-world container (`balena_hello_world` ?), but rename it altogether to something like `balena_healthcheck` or `balena_engine_healthcheck`. It would then become self-explanatory even before customers headed for the docs...

dt-rush, 2019-09-04:
@ZubairLK @pdcastro second (can I second my own feedback?) and I believe this was also Alex's feedback in the all hands

commit e8bdb7da (2019-11-29): balenaEngine: Namespace the hello-world healthcheck image

Fixes #1630

Change-type: minor
Changelog-entry: Namespace the hello-world healthcheck image
commit bc74c5d2 (2019-12-02): balenaEngine: Namespace the hello-world healthcheck image

Fixes #1630

Need to make the script go in the background in ExecStartPost=.
Otherwise, the service status never gets to active/running resulting
it in remaining stuck in an endless loop.

If the health-check load fails for whatever reason, the subsequent
engine healthcheck will fail retriggering the healthcheck load.

Change-type: minor
Changelog-entry: Namespace the hello-world healthcheck image
PR #1764 (2019-12-02): Namespace hello-world healthcheck image
Fixes #1630 

---
### Contributor checklist

- [x] Changes have been tested
- [x] `Change-type` present on at least one commit
- [x] `Signed-off-by` is present
- [x] The PR complies with the [Open Embedded Commit Patch Message Guidelines](http://www.openembedded.org/wiki/Commit_Patch_Message_Guidelines)

### Reviewer Guidelines
- When submitting a review, please pick:
  - '*Approve*' if this change would be acceptable in the codebase (even if there are minor or cosmetic tweaks that could be improved).
  - '*Request Changes*' if this change would not be a [... cut]
commit 4ec47419 (2019-12-23): balenaEngine: Namespace the hello-world healthcheck image

Fixes #1630

Need to make the script go in the background in ExecStartPost=.
Otherwise, the service status never gets to active/running resulting
it in remaining stuck in an endless loop.

If the health-check load fails for whatever reason, the subsequent
engine healthcheck will fail retriggering the healthcheck load.

Change-type: minor
Changelog-entry: Namespace the hello-world healthcheck image
closed 2019-12-02 by ZubairLK