balena-os/meta-balena #1633: Investigate exactly how docker-autoheal breaks the balena socket

ZubairLK, 2019-08-28:
https://github.com/willfarrell/docker-autoheal seems to not play well with balenaOS.

The balena socket in /var/lib/balena.sock turns into a folder!

Investigate how this happens so that we can attempt to improve robustness if possible.

https://www.flowdock.com/app/rulemotion/r-architecture/threads/M8yBO3CLNZgFBfYTt76Fi016eVM

vinicyusmacedo, 2019-10-16:
Hum, I'm interested on taking a look at it.

walpoletim, 2019-10-29:
I have a similar situation where my Balena goat container socket turns into a folder

vinicyusmacedo, 2019-11-03:
Hello,

So, I gave it a try:

- If you run docker with -v mounting an existing file, it won't try to create anything
- If you run it with -v mounting a file that doesn't exist, it will create a directory
- I spin up a dev image of balenaOS on my rpi3b and I couldn't find the /var/lib/balena.sock, only /var/run/balena.sock and /var/run/balena-engine.sock.

@ZubairLK is docker-autoheal provisioned alongside balenaOS? Or did you run it separately?

walpoletim, 2019-11-03:
Check this thread out - It may be related..

https://forums.balena.io/t/raspberry-pi-4-keeps-restarting-upload/36478/109

vinicyusmacedo, 2019-11-03:
@walpoletim I see that this error `failed to delete container from containerd: no such container` pops a lot in your logs. Might be related to the race condition they mention in https://github.com/moby/moby/issues/30348?

Can you spot exactly when the docker daemon restarts (if it does)?