balena-os/meta-balena #1635: Investigate rollback-altboot service udev dependency

ZubairLK, 2019-08-28:
These show up in boot.

```
Aug 08 10:01:27 localhost sh[494]: find: /mnt/sysroot/inactive/: No such device
Aug 08 10:01:27 localhost systemd[1]: Reached target Local File Systems.
Aug 08 10:01:27 localhost sh[494]: /usr/bin/rollback-check-version: line 11: .: : file not found
Aug 08 10:01:27 localhost sh[494]: find: /mnt/sysroot/inactive/: No such device
```
Also see.
https://github.com/balena-os/balena-jetson/commit/5a461ad35280698c8b0f05cb7b2fab7d34e6d9aa

ZubairLK, 2019-08-28:
@floion i'm guessing these are just after HUP?

floion, 2019-08-28:
Both issues were noticed on rpi3 and jetson tx2 which are running on warrior, if that helps to narrow down the problem.

acostach, 2019-08-28:
@ZubairLK on TX2 and RPI3 yes, just after HUP.

ZubairLK, 2019-09-06:
@acostach please move the bsp workarounds now that this is merged.

PR #1643 (2019-09-03): balena-rollback: Fix service dependency
Fixes bug introduced by #1441 
Add systemd-udev-settle.service to the rollback services.

Tested on pi3.
Fixes #1635

---
### Contributor checklist

- [x] Changes have been tested
- [x] `Change-type` present on at least one commit
- [x] `Signed-off-by` is present
- [x] The PR complies with the [Open Embedded Commit Patch Message Guidelines](http://www.openembedded.org/wiki/Commit_Patch_Message_Guidelines)

### Reviewer Guidelines
- When submitting a review, please pick:
  - '*Approve*' if this change would be acceptable in the codebase (even if there are [... cut]
closed 2019-09-06 by ZubairLK