balena-os/meta-balena #1637: Explicitly define to have packages that externally run scripts rely on

imrehg, 2019-08-28:
Discussed in https://github.com/balena-os/resinhup/pull/240 that's it's needed, eg. to ensure that `busybox` compiled with `tee` being available

This effect external scripts such as HUP, and diagnose (which are likely 2 separate changes to this repo, but conceptually coming from the same place.

alexgg, 2020-04-17:
@ZubairLK this definition is unclear to me, maybe you know more about this?

imrehg, 2020-04-17:
@alexgg external scripts use common userspace tools in Linux, but since busybox can be configured to be built without them, should ensure that it is explicitly set to be built with them. Example above is `tee` which is/was used in https://github.com/balena-os/resinhup/blob/eca09e54e4816c448a0e7ef7bc864be060893fad/upgrade-2.x.sh#L758 and thus would break if suddenly the binary wasn't available in the system. Other examples might be from @xginn8 & the diagnostics side.
Does this make some sense? Just to drop a better context, as likely the above one didn't quite did the job. :)

alexgg, 2020-05-04:
thanks @imrehg I understand now.

Code reviews should take this into account and indeed any recipe that includes a script should enforce dependencies on any tool the script uses.
However, I think we can assume that currently these dependencies are met, otherwise stuff will break. We also get some coverage from the release testing.
I don't see an immediate action apart from making sure code reviews take these into account, so I am closing this.

xginn8, 2020-05-04:
@alexgg i think the point of this issue was to explicitly define what we rely on in those sorts of scripts, rather than relying on something failing two or three places removed from the source of the issue. As such, there exists some set of tools (like resinhup) that are run/controlled remotely but rely on on-device binaries, so there is no corresponding recipe. I believe the intent here was to create a dummy recipe outlining those dependencies, to prevent regression.

alexgg, 2020-05-04:
hi @xginn8 the update logic in the proxy should ideally be OS independent. In the near future we will allow for OS customizations (maybe via dockerfiles or other mechanism) and enforcing the type of dependency you describe above won't work. It does sound like some of the resinhup logic actually belongs in the OS itself.
We can leave this open, but I don't think there is enough definition here to schedule any work.