balena-os/meta-balena #1645: Investigate errors `Failed to attach 1 to compat systemd cgroup `

ZubairLK, 2019-08-29:
Check these out

https://forums.balena.io/t/container-reboot-and-failed-to-attach-1-to-compat-systemd-cgroup/27739

https://forums.balena.io/t/container-fails-to-start-failed-to-attach-1-to-compat-systemd-cgroup/28537

ZubairLK, 2019-09-03:
reproduced.
systemd + privileged container, pi3 v2.41. restart balena service twice.

More detail on the forum

ZubairLK, 2019-09-03:
reproduced on nuc as well. 2.41.

ZubairLK, 2019-09-05:
Reproduced on pi3 v2.38 running systemd 239 on host and running systemd 241 inside privileged container.

How/why the issue manifests now is a separate question that I haven't answered yet?

But I think the issue is that we run balena-engine with the cgroupfs driver
```
root@222b941:~# balena info | grep -i cgroupfs
Cgroup Driver: cgroupfs
```

While systemd will be trying to use some of its own magic to manage cgroups.
Adding the following to `/etc/docker/daemon.json` and restarting balena-engine doesn't make the problem happen (at least during my limited testing)
```
  "exec-opts": ["native.cgroupdriver=systemd"]
```

This test was based on https://github.com/kubernetes/kubeadm/issues/1394 and specifically  https://github.com/kubernetes/kubeadm/issues/1394#issuecomment-462878219

alsmith, 2019-09-05:
I didn't have a daemon.json so I did this and rebooted the system...

root@468439d:~# cat /etc/docker/daemon.json 
{ "exec-opts": ["native.cgroupdriver=systemd"] }
root@468439d:~#

ZubairLK, 2019-09-05:
v2.31.5 systemd 237 on host.
buster in app container (privileged, systemd 241) reproduces the issue.
just stopping/starting container is enough to trigger it

ZubairLK, 2019-09-06:
Adding `systemd.legacy_systemd_cgroup_controller` to the container entry.sh script to make systemd start with that arg also makes the problem go away..
here https://github.com/balena-io-playground/balenalib-systemd-example/blob/master/app/systemd/entry.sh#L88

ZubairLK, 2019-09-09:
We have a way forward now.
Will be setting `systemd` as the cgroup driver in future os versions.
For devices running the previous hostOS running new base images, we can suggest the `systemd.legacy_systemd_cgroup_controller` workaround.

ZubairLK, 2019-09-09:
https://www.flowdock.com/app/rulemotion/r-resinos/threads/KrRaUB4YA56p79NtNbHYktVVBKS

PR #1658 (2019-09-10): balenaEngine: Switch cgroup driver to systemd
The default driver is cgroupfs. We switch to systemd so that there is
one cgroup manager in our OS.

Otherwise, systemd will have its own cgroup manager and cgroupfs will
be another cgroup manager via balenaEngine daemon.

Fixes #1645 which contains more information

Change-type: minor
Changelog-entry: Fix a bug where application containers with new systemd versions were failing to start in cases. Switch to systemd cgroup driver in balenaEngine

---
### Contributor checklist

- [x] Changes have been tested
- [x] `Change-type` present on at least on [... cut]
commit a79f18a3 (2019-09-10): balenaEngine: Switch cgroup driver to systemd

The default driver is cgroupfs. We switch to systemd so that there is
one cgroup manager in our OS.

Otherwise, systemd will have its own cgroup manager and cgroupfs will
be another cgroup manager via balenaEngine daemon.

Fixes #1645 which contains more information

Change-type: minor
Changelog-entry: Fix a bug where application containers with new systemd versions were failing to start in cases. Switch to systemd cgroup driver in balenaEngine
closed 2019-09-13 by ZubairLK