balena-os/meta-balena #1651: Openvpn was dead on device, and didn't restart

imrehg, 2019-08-30:
balenaOS 2.31.5+rev1, Fin. Marked offline for 4 hours, but still logs. Another device on the same network was available, and could connect across. The openvpn service was dead, but logs are rotated and we couldn't gather any info

```
# systemctl status openvpn
● openvpn.service - OpenVPN
   Loaded: loaded (/lib/systemd/system/openvpn.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Fri 2019-08-30 12:13:47 UTC; 4h 39min ago
  Process: 2696 ExecStart=/usr/sbin/openvpn --writepid /var/run/openvpn/openvpn.pid --cd /etc/openvpn/ --config /etc/openvpn/openvpn.conf --connect-retry 5 120 (code=exited, status=0/SUCCESS)
 Main PID: 2696 (code=exited, status=0/SUCCESS)

Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable.
```
No other service was failed

ZubairLK, 2019-09-02:
@imrehg this is strange. The service is set to restart. Wonder why systemd decided not to do that..

```
zubairlk@zubair-xps-resin:~/resin/yocto/resin-intel$ docker run -v /usr/bin/qemu-arm-static:/usr/bin/qemu-arm-static  --rm -i -t resin/resinos:2.31.5_rev1.dev-fincm3 /bin/bash
bash-4.4# cat /lib/systemd/system/openvpn.service
[Unit]
Description=OpenVPN
Requires=prepare-openvpn.service bind-etc-openvpn.service
After=syslog.target network.target prepare-openvpn.service bind-etc-openvpn.service time-sync.target
ConditionFileNotEmpty=/etc/openvpn/openvpn.conf

[Service]
PrivateTmp=true
Restart=always
RestartSec=10s
#Adjust OOMscore to -1000 to disable OOM killing for openvpn
OOMScoreAdjust=-1000
PIDFile=/var/run/openvpn/openvpn.pid
ExecStart=/usr/sbin/openvpn --writepid /var/run/openvpn/openvpn.pid --cd /etc/openvpn/ --config /etc/openvpn/openvpn.conf --connect-retry 5 120

[Install]
Alias=openvpn-resin.service
WantedBy=multi-user.target
bash-4.4# 
```

ZubairLK, 2019-09-06:
From https://www.freedesktop.org/software/systemd/man/systemd.service.html
> Restart=
Configures whether the service shall be restarted when the service process exits, is killed, or a timeout is reached. The service process may be the main service process, but it may also be one of the processes specified with ExecStartPre=, ExecStartPost=, ExecStop=, ExecStopPost=, or ExecReload=. When the death of the process is a result of systemd operation (e.g. service stop or restart), the service will not be restarted. Timeouts include missing the watchdog "keep-alive ping" deadline and a service start, reload, and stop operation timeouts.

So unless systemd was asked to disable the vpn service. The openvpn service should have restarted..

The `(code=exited, status=0/SUCCESS)` is quite suspicious as the daemon should never have just exited..

Will close this issue for now unless we see it again.
The device was in that state momentarily and restarting the service recovered it.

closed 2019-09-06 by ZubairLK