balena-os/meta-balena #1653: connectivity: device does not reconnect after router reboot

xginn8, 2019-09-02:
Seen on both NUC VM (balenaOS 2.38.3+rev5) & NUC (balenaOS 2.38.3+rev4).

After a router reboot (performed by toggling the `Hotspot` function), devices reconnect automatically but are not routable to the WAN. Specifically, these devices are tethering from an Android moto x4, Android 9. They are locally ssh'able, but unable to connect outside the LAN.

Upon inspection, it seems that the gateway IP address has changed, but NM is unaware of that until requesting a new DHCP lease.

Broken device:
```
root@e655ba7:~# ip route
default via 192.168.43.144 dev wlp58s0  metric 600 
10.114.101.0/24 dev balena0 scope link  src 10.114.101.1 
10.114.102.0/24 dev resin-dns scope link  src 10.114.102.1 
10.114.104.0/25 dev supervisor0 scope link  src 10.114.104.1 
52.4.252.97 dev resin-vpn scope link  src 100.64.42.128 
172.17.0.0/16 dev br-06027e09887f scope link  src 172.17.0.1 
192.168.43.0/24 dev wlp58s0 scope link  src 192.168.43.114  metric 600 
root@e655ba7:~# 
```

Functioning device:
```
root@11f3957:~# ip route
default via 192.168.43.226 dev enp0s3  metric 100 
10.114.101.0/24 dev balena0 scope link  src 10.114.101.1 
10.114.102.0/24 dev resin-dns scope link  src 10.114.102.1 
10.114.104.0/25 dev supervisor0 scope link  src 10.114.104.1 
52.4.252.97 dev resin-vpn scope link  src 100.64.11.202 
172.17.0.0/16 dev br-b831b6806d74 scope link  src 172.17.0.1 
192.168.43.0/24 dev enp0s3 scope link  src 192.168.43.113  metric 100 
root@11f3957:~# 
```
cc @ZubairLK

ZubairLK, 2019-09-03:
key thing to note that other devices on the network managed to pick the new gateway. Arch linux laptop, windows etc.
But the balenaOS device didn't pick it up. I suspect an up/down/state change script is missing for Network Manager and this needs looking into

ZubairLK, 2019-09-16:
https://www.flowdock.com/app/rulemotion/r-resinos/threads/HqP3-NpwCh73vX-awBwJO7Ee_2E?filter=search&query=xginn8

ZubairLK, 2019-10-08:
1

ZubairLK, 2019-10-09:
Did some further digging. We have an issue I think. On connection change, the old ip routes are not removed/updated
Snippet from raspbian for reference.
```
-- Logs begin at Wed 2019-10-09 17:37:05 BST. --
Oct 09 17:44:49 raspberrypi dhcpcd[521]: wlan0: carrier lost
Oct 09 17:44:49 raspberrypi dhcpcd[521]: wlan0: deleting address fe80::7e83:767f:cb24:736c
Oct 09 17:44:49 raspberrypi avahi-daemon[342]: Withdrawing address record for fe80::7e83:767f:cb24:736c on wlan0.
Oct 09 17:44:49 raspberrypi avahi-daemon[342]: Leaving mDNS multicast group on interface wlan0.IPv6 with address fe80::7e83:767f:cb24:736c.
Oct 09 17:44:49 raspberrypi avahi-daemon[342]: Interface wlan0.IPv6 no longer relevant for mDNS.
Oct 09 17:44:49 raspberrypi avahi-daemon[342]: Withdrawing address record for 192.168.43.243 on wlan0.
Oct 09 17:44:49 raspberrypi avahi-daemon[342]: Leaving mDNS multicast group on interface wlan0.IPv4 with address 192.168.43.243.
Oct 09 17:44:49 raspberrypi avahi-daemon[342]: Interface wlan0.IPv4 no longer relevant for mDNS.
Oct 09 17:44:49 raspberrypi dhcpcd[521]: wlan0: deleting route to 192.168.43.0/24
Oct 09 17:44:49 raspberrypi dhcpcd[521]: wlan0: deleting default route via 192.168.43.45
Oct 09 17:44:54 raspberrypi dhcpcd[521]: wlan0: carrier acquired
Oct 09 17:44:54 raspberrypi dhcpcd[521]: wlan0: IAID 32:06:7f:ba
Oct 09 17:44:54 raspberrypi dhcpcd[521]: wlan0: adding address fe80::7e83:767f:cb24:736c
Oct 09 17:44:54 raspberrypi avahi-daemon[342]: Joining mDNS multicast group on interface wlan0.IPv6 with address fe80::7e83:767f:cb24:736c.
Oct 09 17:44:54 raspberrypi avahi-daemon[342]: New relevant interface wlan0.IPv6 for mDNS.
Oct 09 17:44:54 raspberrypi avahi-daemon[342]: Registering new address record for fe80::7e83:767f:cb24:736c on wlan0.*.
Oct 09 17:44:55 raspberrypi dhcpcd[521]: wlan0: soliciting an IPv6 router
Oct 09 17:44:55 raspberrypi dhcpcd[521]: wlan0: rebinding lease of 192.168.43.243
Oct 09 17:44:55 raspberrypi dhcpcd[521]: wlan0: probing address 192.168.43.243/24
Oct 09 17:45:00 raspberrypi dhcpcd[521]: wlan0: leased 192.168.43.243 for 3599 seconds
Oct 09 17:45:00 raspberrypi avahi-daemon[342]: Joining mDNS multicast group on interface wlan0.IPv4 with address 192.168.43.243.
Oct 09 17:45:00 raspberrypi avahi-daemon[342]: New relevant interface wlan0.IPv4 for mDNS.
Oct 09 17:45:00 raspberrypi avahi-daemon[342]: Registering new address record for 192.168.43.243 on wlan0.IPv4.
Oct 09 17:45:00 raspberrypi dhcpcd[521]: wlan0: adding route to 192.168.43.0/24
Oct 09 17:45:00 raspberrypi dhcpcd[521]: wlan0: adding default route via 192.168.43.123
```

ZubairLK, 2019-10-10:
Ok. I think I have a handle on this one now.
We use the NetworkManagers internal dhcp client. Raspbian uses `dhcpcd`. My laptop uses `dhclient`

What I suspect is that `dhcpcd`/`dhclient` are better at handling such sneaky updates to gateway ips of the hotspot when switching on/off.

But NetworkManagers internal dhcp client isn't as good. However, there is an update to this. v2.43+ has NetworkManager 1.20+. And I can't reproduce this issue easily on that version.

I see that when I toggle the hotspot on my phone, the routes in balenaOS on a pi3 running 2.43 do get updated. Whereas on v2.38, the routes would remain the same old ones.

So there has been some improvement in NetworkManager between 1.18 and 1.20.

It is still flaky. Sometimes I see routes update. Sometimes I don't. Ping somehow still works.

 So @xginn8 can you see if you can replicate this on the nuc v2.43+?