balena-os/meta-balena #1671: udev rule refers to binary that doesn't exist on the system

imrehg, 2019-09-16:
`/lib/udev/rules.d/97-hid2hci.rules` refers to `hid2hci` binary, which doesn't exist on balenaOS. Thus if there's a device that would trigger that rule, `systemd-udevd` goes into a busy-loop, eating up 70+% percent of the CPU, and not finishing ever.

```
Sep 16 14:05:41 balena systemd-udevd[531]: Process 'hid2hci --method=dell --devpath=/devices/pci0000:00/0000:00:1a.0/usb3/3-1/3-1.2/3-1.2:1.0' failed with exit code 1.
```

We should not have rules that refer to binaries that doesn't exist on the system.

[Flowdock discusssion](https://www.flowdock.com/app/rulemotion/r-beginners/threads/cEbys4fkDmHwXAvdRKRMbhJbbIY)

xginn8, 2019-09-16:
Related to https://github.com/balena-os/meta-balena/issues/1637

ZubairLK, 2019-09-23:
@xginn8 @imrehg what os version was this?
I checked a few and I do find the binary in this path `/lib/udev/hid2hci`

ZubairLK, 2019-09-23:
was it an old dell laptop?
Could be something this then..
https://bugs.launchpad.net/ubuntu/+source/systemd/+bug/1759836

rahul-thakoor, 2019-09-23:
indeed @ZubairLK it's an Inspiron 1545

ZubairLK, 2019-09-24:
@rahul-thakoor so the original issue suggests a missing binary which isn't the case.
I'd say have a look at https://bugs.launchpad.net/ubuntu/+source/systemd/+bug/1759836 to see if you can narrow down the issue. Its a hardware specific one I think and not a generic balenaOS thing.

ZubairLK, 2019-10-15:
Closing as I don't think it is a balenaOS issue.

The original issue suggests the binary does not exist on the system. But it does.

closed 2019-10-15 by ZubairLK