balena-os/meta-balena #1673: kernel-module-build example fails for jetson-nano

acostach, 2019-09-17:
BalenaOS version is:  2.43.0

Failure occurs in build.sh line 127

make: Entering directory '/tmp/tmp.jKWHC9HXZG'
make: *** No rule to make target 'modules_prepare'.  Stop.
make: Leaving directory '/tmp/tmp.jKWHC9HXZG'

root@04456ca:/usr/src/app# ls /tmp/tmp.jKWHC9HXZG/
build  kernel_source.tar.gz  source

root@04456ca:/usr/src/app/example_module# make -C /tmp/tmp.jKWHC9HXZG/build/
make: Entering directory '/tmp/tmp.jKWHC9HXZG/build'
  CHK     include/config/kernel.release
  CHK     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/basic/bin2c
make[1]: *** No rule to make target 'arch/arm64/kernel/vdso/vdso.lds', needed by 'arch/arm64/kernel/vdso/vdso.so.dbg'.  Stop.
make: *** [arch/arm64/Makefile:176: vdso_prepare] Error 2
make: Leaving directory '/tmp/tmp.jKWHC9HXZG/build'

# Kernel version is
root@04456ca:/usr/src/app/example_module# uname -a
Linux 04456ca 4.9.140-l4t-r32.2+g3dcbed5 #1 SMP PREEMPT Mon Sep 16 12:44:59 UTC 2019 aarch64 GNU/Linux

# Headers version matches: 4.9.140-l4t-r32.2+g3dcbed5
root@04456ca:/tmp/tmp.jKWHC9HXZG/source# ls
Documentation  Module.symvers                         block   drivers   include  kernel  modules.order  samples   sound  virt
Kconfig        System.map-4.9.140-l4t-r32.2+g3dcbed5  certs   firmware  init     lib     net            scripts   tools
Makefile       arch                                   crypto  fs        ipc      mm      nvidia         security  usr

agherzan, 2019-09-17:
I'm pretty sure I've need the fix for this in master `meta-balena`. @ZubairLK to confirm.

ZubairLK, 2019-09-17:
@acostach can you check https://github.com/balena-io-projects/kernel-module-build/pull/28

acostach, 2019-09-25:
@ZubairLK I confirm that with the latest fixes in the kernel-module-build at v1.0.7, it builds fine.

ZubairLK, 2019-09-26:
can't reproduce so closing this again..

closed 2019-09-26 by ZubairLK