balena-os/meta-balena #1674: Unable to mount HFS in latest image (intel atom)

ned-kelly, 2019-09-17:
Not seeing HFS support in `/proc/filesystems`

Any tips to mounting HFS formatted drives inside a container?

```
root@3c2cfc5fa80a:/media# apt install hfsplus

root@3c2cfc5fa80a:/media# mount -t auto -o sync,noatime /dev/sda1 /media/hfsdrive
mount: unknown filesystem type 'hfsplus'
```

ZubairLK, 2019-09-19:
@ned-kelly probably needs a kernel module that we don't enable. Could you please elaborate if you have a particular reason you are looking for HFS on a balenaOS device?

ned-kelly, 2019-09-19:
> @ned-kelly probably needs a kernel module that we don't enable. Could you please elaborate if you have a particular reason you are looking for HFS on a balenaOS device?

Yep, we need to support mounting filesystems from OSX - Namely large drives that contain content we ingest (via Balena) from end users (image training libraries for Tensorflow etc)  - Don't think this should be enabled on all OS's however if we could at least get it enabled on the Intel based builds that might make sense - Since if you're using an Intel based device, it's probably a device with a bit more grunt that might be doing more complex tasks then a simple ARM based SBC deployment.

floion, 2019-09-24:
This should be done by https://github.com/balena-os/balena-intel/pull/252

closed 2019-09-24 by ZubairLK