balena-os/meta-balena #1683: Device always shows as offline when in local mode

krasi-georgiev, 2019-09-23:
The device shows as offline all the time in the balena portal. If I switch to non local mode it shows as online. Otherwise I can ssh to it locally and 
`curl https://api.resin.io/ping` returns OK

this is Rpi 3b+ , balenaOS 2.38.0+rev1	
I can enable support access if needed.
 
here are the logs after running
```
systemctl restart openvpn-resin && systemctl restart resin-supervisor
```

```

Sep 23 19:57:17 93ae850 systemd[1]: Starting OpenSSH Per-Connection Daemon (192.168.0.102:35368)...
Sep 23 19:57:17 93ae850 systemd[1]: Started OpenSSH Per-Connection Daemon (192.168.0.102:35368).
Sep 23 19:58:41 93ae850 systemd[1]: Starting Prepare OpenVPN...
Sep 23 19:58:41 93ae850 prepare-openvpn[30788]: prepare-openvpn: [INFO] Resin.io VPN authentication.
Sep 23 19:58:41 93ae850 systemd[1]: Started Prepare OpenVPN.
Sep 23 19:58:41 93ae850 systemd[1]: Started OpenVPN.
Sep 23 19:58:41 93ae850 openvpn[30792]: Mon Sep 23 19:58:41 2019 WARNING: file '/var/volatile/vpn-auth' is group or others accessible
Sep 23 19:58:41 93ae850 openvpn[30792]: Mon Sep 23 19:58:41 2019 OpenVPN 2.4.3 arm-poky-linux-gnueabi [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD] built on May 13 2019
Sep 23 19:58:41 93ae850 openvpn[30792]: Mon Sep 23 19:58:41 2019 library versions: OpenSSL 1.1.1a  20 Nov 2018, LZO 2.10
Sep 23 19:58:41 93ae850 openvpn[30792]: Mon Sep 23 19:58:41 2019 NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Sep 23 19:58:41 93ae850 systemd[1]: Stopping Resin supervisor...
Sep 23 19:58:41 93ae850 openvpn[30792]: Mon Sep 23 19:58:41 2019 TCP/UDP: Preserving recently used remote address: [AF_INET]18.232.192.190:443
Sep 23 19:58:41 93ae850 openvpn[30792]: Mon Sep 23 19:58:41 2019 Socket Buffers: R=[87380->87380] S=[16384->16384]
Sep 23 19:58:41 93ae850 openvpn[30792]: Mon Sep 23 19:58:41 2019 Attempting to establish TCP connection with [AF_INET]18.232.192.190:443 [nonblock]
Sep 23 19:58:42 93ae850 openvpn[30792]: Mon Sep 23 19:58:42 2019 TCP connection established with [AF_INET]18.232.192.190:443
Sep 23 19:58:42 93ae850 openvpn[30792]: Mon Sep 23 19:58:42 2019 TCP_CLIENT link local: (not bound)
Sep 23 19:58:42 93ae850 openvpn[30792]: Mon Sep 23 19:58:42 2019 TCP_CLIENT link remote: [AF_INET]18.232.192.190:443
Sep 23 19:58:42 93ae850 openvpn[30792]: Mon Sep 23 19:58:42 2019 NOTE: UID/GID downgrade will be delayed because of --client, --pull, or --up-delay
Sep 23 19:58:42 93ae850 openvpn[30792]: Mon Sep 23 19:58:42 2019 TLS: Initial packet from [AF_INET]18.232.192.190:443, sid=6be56de1 d1205562
Sep 23 19:58:42 93ae850 openvpn[30792]: Mon Sep 23 19:58:42 2019 WARNING: this configuration may cache passwords in memory -- use the auth-nocache option to prevent this
Sep 23 19:58:42 93ae850 openvpn[30792]: Mon Sep 23 19:58:42 2019 VERIFY OK: depth=1, C=AU, ST=Some-State, O=Internet Widgits Pty Ltd
Sep 23 19:58:42 93ae850 openvpn[30792]: Mon Sep 23 19:58:42 2019 VERIFY KU OK
Sep 23 19:58:42 93ae850 openvpn[30792]: Mon Sep 23 19:58:42 2019 Validating certificate extended key usage
Sep 23 19:58:42 93ae850 openvpn[30792]: Mon Sep 23 19:58:42 2019 ++ Certificate has EKU (str) TLS Web Server Authentication, expects TLS Web Server Authentication
Sep 23 19:58:42 93ae850 openvpn[30792]: Mon Sep 23 19:58:42 2019 VERIFY EKU OK
Sep 23 19:58:42 93ae850 openvpn[30792]: Mon Sep 23 19:58:42 2019 VERIFY OK: depth=0, C=US, ST=WA, L=Seattle, O=Resin.io, OU=VPN, CN=ResinVPN
Sep 23 19:58:43 93ae850 openvpn[30792]: Mon Sep 23 19:58:43 2019 Control Channel: TLSv1.3, cipher TLSv1.3 TLS_AES_256_GCM_SHA384, 2048 bit RSA
Sep 23 19:58:43 93ae850 openvpn[30792]: Mon Sep 23 19:58:43 2019 [ResinVPN] Peer Connection Initiated with [AF_INET]18.232.192.190:443
Sep 23 19:58:44 93ae850 openvpn[30792]: Mon Sep 23 19:58:44 2019 SENT CONTROL [ResinVPN]: 'PUSH_REQUEST' (status=1)
Sep 23 19:58:44 93ae850 openvpn[30792]: Mon Sep 23 19:58:44 2019 PUSH: Received control message: 'PUSH_REPLY,sndbuf 0,rcvbuf 0,route 52.4.252.97,ping 10,ping-restart 60,socket-flags TCP_NODELAY,ifconfig 10.240.27.73 52.4.252.97,peer-id 0,cipher AES-256-GCM'
Sep 23 19:58:44 93ae850 openvpn[30792]: Mon Sep 23 19:58:44 2019 OPTIONS IMPORT: timers and/or timeouts modified
Sep 23 19:58:44 93ae850 openvpn[30792]: Mon Sep 23 19:58:44 2019 OPTIONS IMPORT: --sndbuf/--rcvbuf options modified
Sep 23 19:58:44 93ae850 openvpn[30792]: Mon Sep 23 19:58:44 2019 Socket Buffers: R=[341760->341760] S=[44800->44800]
Sep 23 19:58:44 93ae850 openvpn[30792]: Mon Sep 23 19:58:44 2019 OPTIONS IMPORT: --socket-flags option modified
Sep 23 19:58:44 93ae850 openvpn[30792]: Mon Sep 23 19:58:44 2019 Socket flags: TCP_NODELAY=1 succeeded
Sep 23 19:58:44 93ae850 openvpn[30792]: Mon Sep 23 19:58:44 2019 OPTIONS IMPORT: --ifconfig/up options modified
Sep 23 19:58:44 93ae850 openvpn[30792]: Mon Sep 23 19:58:44 2019 OPTIONS IMPORT: route options modified
Sep 23 19:58:44 93ae850 openvpn[30792]: Mon Sep 23 19:58:44 2019 OPTIONS IMPORT: peer-id set
Sep 23 19:58:44 93ae850 openvpn[30792]: Mon Sep 23 19:58:44 2019 OPTIONS IMPORT: adjusting link_mtu to 1627
Sep 23 19:58:44 93ae850 openvpn[30792]: Mon Sep 23 19:58:44 2019 OPTIONS IMPORT: data channel crypto options modified
Sep 23 19:58:44 93ae850 openvpn[30792]: Mon Sep 23 19:58:44 2019 Data Channel: using negotiated cipher 'AES-256-GCM'
Sep 23 19:58:44 93ae850 openvpn[30792]: Mon Sep 23 19:58:44 2019 Data Channel Encrypt: Cipher 'AES-256-GCM' initialized with 256 bit key
Sep 23 19:58:44 93ae850 openvpn[30792]: Mon Sep 23 19:58:44 2019 Data Channel Decrypt: Cipher 'AES-256-GCM' initialized with 256 bit key
Sep 23 19:58:44 93ae850 openvpn[30792]: Mon Sep 23 19:58:44 2019 ROUTE_GATEWAY 192.168.0.1/255.255.255.0 IFACE=wlan0 HWADDR=b8:27:eb:4b:e7:ac
Sep 23 19:58:44 93ae850 openvpn[30792]: Mon Sep 23 19:58:44 2019 TUN/TAP device resin-vpn opened
Sep 23 19:58:44 93ae850 openvpn[30792]: Mon Sep 23 19:58:44 2019 TUN/TAP TX queue length set to 100
Sep 23 19:58:44 93ae850 openvpn[30792]: Mon Sep 23 19:58:44 2019 do_ifconfig, tt->did_ifconfig_ipv6_setup=0
Sep 23 19:58:44 93ae850 openvpn[30792]: Mon Sep 23 19:58:44 2019 /sbin/ip link set dev resin-vpn up mtu 1500
Sep 23 19:58:44 93ae850 openvpn[30792]: Mon Sep 23 19:58:44 2019 /sbin/ip addr add dev resin-vpn local 10.240.27.73 peer 52.4.252.97
Sep 23 19:58:44 93ae850 systemd-udevd[30811]: link_config: autonegotiation is unset or enabled, the speed and duplex are not writable.
Sep 23 19:58:44 93ae850 openvpn[30792]: Mon Sep 23 19:58:44 2019 /etc/openvpn-misc/upscript.sh resin-vpn 1500 1555 10.240.27.73 52.4.252.97 init
Sep 23 19:58:44 93ae850 NetworkManager[690]: <info>  [1569268724.3648] manager: (resin-vpn): new Tun device (/org/freedesktop/NetworkManager/Devices/806)
Sep 23 19:58:44 93ae850 openvpn[30792]: resin-ntp-config: Found config.json in /mnt/boot/config.json .
Sep 23 19:58:44 93ae850 openvpn[30792]: Mon Sep 23 19:58:44 2019 /sbin/ip route add 52.4.252.97/32 via 52.4.252.97
Sep 23 19:58:44 93ae850 openvpn[30792]: ip: RTNETLINK answers: File exists
Sep 23 19:58:44 93ae850 openvpn[30792]: Mon Sep 23 19:58:44 2019 ERROR: Linux route add command failed: external program exited with error status: 2
Sep 23 19:58:44 93ae850 openvpn[30792]: Mon Sep 23 19:58:44 2019 GID set to openvpn
[... 56 lines cut ...]
Sep 23 19:59:00 93ae850 resin-supervisor[30862]: Finished applying target state
Sep 23 19:59:00 93ae850 resin-supervisor[30862]: Apply success!
Sep 23 19:59:00 93ae850 87df77aa89b4[26543]: Apply success!
Sep 23 19:59:01 93ae850 87df77aa89b4[26543]: Applying target state
Sep 23 19:59:01 93ae850 resin-supervisor[30862]: Applying target state
Sep 23 19:59:01 93ae850 87df77aa89b4[26543]: Event: Apply config change in progress {}
Sep 23 19:59:01 93ae850 resin-supervisor[30862]: Event: Apply config change in progress {}
Sep 23 19:59:01 93ae850 openvpn[30792]: Mon Sep 23 19:59:01 2019 event_wait : Interrupted system call (code=4)
Sep 23 19:59:01 93ae850 openvpn[30792]: Mon Sep 23 19:59:01 2019 Closing TUN/TAP interface
Sep 23 19:59:01 93ae850 openvpn[30792]: Mon Sep 23 19:59:01 2019 /sbin/ip addr del dev resin-vpn local 10.240.27.73 peer 52.4.252.97
Sep 23 19:59:01 93ae850 systemd[1]: Stopping OpenVPN...
Sep 23 19:59:01 93ae850 resin-supervisor[30862]: Event: Apply config change success {}
Sep 23 19:59:01 93ae850 87df77aa89b4[26543]: Event: Apply config change success {}
Sep 23 19:59:01 93ae850 openvpn[30792]: ip: RTNETLINK answers: Operation not permitted
[... cut ...]
Sep 23 19:59:02 93ae850 openvpn[30792]: Mon Sep 23 19:59:02 2019 SIGTERM[hard,] received, process exiting
Sep 23 19:59:02 93ae850 systemd[1]: Stopped OpenVPN.
Sep 23 19:59:02 93ae850 87df77aa89b4[26543]: Finished applying target state
Sep 23 19:59:02 93ae850 87df77aa89b4[26543]: Apply success!
Sep 23 19:59:02 93ae850 resin-supervisor[30862]: Finished applying target state
Sep 23 19:59:02 93ae850 resin-supervisor[30862]: Apply success!
Sep 23 20:00:44 93ae850 wpa_supplicant[767]: wlan0: WPA: Group rekeying completed with 68:ff:7b:3e:cd:76 [GTK=CCMP]

```

ZubairLK, 2019-09-24:
@krasi-georgiev 
```
Sep 23 19:59:01 93ae850 systemd[1]: Stopping OpenVPN...
```

The line above is quite suspicious. As if something is asking the vpn to switch off?
You haven't disabled vpn via the dashboard by any chance? Or tweaked the `RESIN_SUPERVISOR_VPN_CONTROL` variable..

closed 2019-09-24 by krasi-georgiev

krasi-georgiev, 2019-09-24:
Just double checked and you are right. I must have pressed it by mistake.
Sorry for the noise!