balena-os/meta-balena #1699: Remove double copies of /boot files

ZubairLK, 2019-10-08:
Files in `/boot` are redundant and take up unnecessary space/bandwidth.
Files in `/mnt/sysroot/active/current/boot` are the real ones.

The `/boot` we see in the rootfs while running balenaOS is inside the rootfs container.
Not what the bootloader sees.

(also people get caught changing the kernel there quite a few times and then realising the actual kernel to change is in `/mnt/sysroot/active/current/boot/`)

ZubairLK, 2019-10-08:
side note `/resin-boot` are needed and used by hooks/rollbacks. Even though there is an extra copy in the boot partition and in the root partition.

alexgg, 2020-06-19:
For a raspberrypi4:
```
root@localhost:~# ls /boot/
init  storage-driver  u-boot-raspberrypi3-2019.01-r0.bin  u-boot.bin  zImage

root@localhost:~# ls /resin-boot/
bcm2708-rpi-b-plus.dtb  bcm2710-rpi-3-b-plus.dtb          boot.scr      device-type.json    kernel7.img          splash
bcm2708-rpi-b.dtb       bcm2710-rpi-3-b.dtb               bootcode.bin  fixup.dat           os-release           start.elf
bcm2708-rpi-cm.dtb      bcm2710-rpi-cm3.dtb               cmdline.txt   fixup_cd.dat        overlays             start_cd.elf
bcm2708-rpi-zero-w.dtb  bcm2711-rpi-4-b.dtb               config.json   fixup_x.dat         resin-image          start_x.elf
bcm2709-rpi-2-b.dtb     bcm2835-bootfiles-20200220.stamp  config.txt    image-version-info  resinos.fingerprint  system-connections
```

From there, `init` and `storage-driver` are needed by `hostapp-update`.  `zImage` is actually used by the bootloader to boot the kernel. The only ones that are unused as the device boots from `resin-boot` are:
* u-boot-raspberrypi3-2019.01-r0.bin
* u-boot.bin

On an Intel-NUC:

```
root@b0f8b20:~# ls /boot
EFI  bzImage  init  storage-driver

root@b0f8b20:~# ls /resin-boot
EFI  config.json  device-type.json  grub  image-version-info  os-release  resin-image  resinos.fingerprint  splash  system-connections
```

In this case the EFI folder is unused. This is around 600K of storage.