balena-os/meta-balena #1721: Enable chrony NTP server functionality

imrehg, 2019-10-29:
Have a way to enable NTP server functionality of `chrony` which might be just adding the `allow` keyword to `/etc/chrony.conf` See more at https://chrony.tuxfamily.org/doc/3.4/chrony.conf.html#_ntp_server

This would allow other balenaOS devices on the same network to use one device as an NTP server, as it might be useful in a proxied environment.

I wonder if just enabling it by default is a good thing (or it's a potential liability), or enabling it with a `config.json` setting on demand)

[Flowdock discussion](https://www.flowdock.com/app/rulemotion/public-s-premium/threads/SPCqBIHbTE2sIDKK8Yg7fysTOJs)

ZubairLK, 2019-11-19:
Closing as internal discussion led to the realisation that chrony can be started as a server inside the container. Nothing to do for the host.

closed 2019-11-19 by ZubairLK

imrehg, 2019-11-19:
That's totally fair, and was done in a project indeed :) Duh!