balena-os/meta-balena #1745: Healthdog 0.1.0 fails to be compiled with the latest meta-rust

kbumsik, 2019-11-13:
After I updated all meta layers (including meta-balena v2.45.0) healthdog begin to fail to be compiled.

I guess it somehow have a problem with the latest meta-rust https://github.com/meta-rust/meta-rust/commit/5cda04c7c2158fc982502ca1e7893f71e4ca1729. It compiles fine when I switch it back to https://github.com/meta-rust/meta-rust/commit/84710b341be53cc2af13ecc8ae87773cf84687bf, which is the same as what balena-raspberrypi uses.

```
DEBUG: Executing shell function do_compile
NOTE: cargo = /yocto/work/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/healthdog/0.1.0.AUTOINC+a10155844f-r0/recipe-sysroot-native/usr/bin/cargo
NOTE: rustc = 
NOTE: cargo build -v --target arm-poky-linux-gnueabi --release 
       Fresh semver-parser v0.7.0
       Fresh pkg-config v0.3.9
       Fresh libc v0.2.24
       Fresh utf8-cstr v0.1.6
       Fresh bitflags v0.7.0
       Fresh log v0.3.8
       Fresh cfg-if v0.1.1
       Fresh void v1.0.2
       Fresh getopts v0.2.14
       Fresh semver v0.6.0
       Fresh errno v0.2.3
       Fresh nix v0.8.1
       Fresh rustc_version v0.2.1
       Fresh exec v0.3.1
       Fresh libsystemd-sys v0.1.0
   Compiling mbox v0.3.1
     Running `rustc --crate-name mbox /yocto/work/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/healthdog/0.1.0.AUTOINC+a10155844f-r0/cargo_home/bitbake/mbox-0.3.1/src/lib.rs --color never --crate-type lib --emit=dep-info,metadata,link -C opt-level=3 --cfg 'feature="default"' --cfg 'feature="std"' -C metadata=5904261533b651af -C extra-filename=-5904261533b651af --out-dir /yocto/work/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/healthdog/0.1.0.AUTOINC+a10155844f-r0/git/target/arm-poky-linux-gnueabi/release/deps --target arm-poky-linux-gnueabi -C linker=/yocto/work/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/healthdog/0.1.0.AUTOINC+a10155844f-r0/wrapper/target-rust-ccld -L dependency=/yocto/work/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/healthdog/0.1.0.AUTOINC+a10155844f-r0/git/target/arm-poky-linux-gnueabi/release/deps -L dependency=/yocto/work/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/healthdog/0.1.0.AUTOINC+a10155844f-r0/git/target/release/deps --extern libc=/yocto/work/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/healthdog/0.1.0.AUTOINC+a10155844f-r0/git/target/arm-poky-linux-gnueabi/release/deps/liblibc-cc88c750fa2562e8.rlib --cap-lints allow -L /yocto/work/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/healthdog/0.1.0.AUTOINC+a10155844f-r0/recipe-sysroot/usr/lib/rust --remap-path-prefix=/yocto/work/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/healthdog/0.1.0.AUTOINC+a10155844f-r0=/usr/src/debug/healthdog/0.1.0.AUTOINC+a10155844f-r0 -C prefer-dynamic -C 'link-args=-Wl,-rpath=$ORIGIN/../lib/rust:$ORIGIN/../lib/arm-poky-linux-gnueabi/rustlib/arm-poky-linux-gnueabi/lib' --cfg stable_channel`
error: expected one of `!`, `.`, `::`, `;`, `?`, `{`, `}`, or an operator, found `<-`
   --> /usr/src/debug/healthdog/0.1.0.AUTOINC+a10155844f-r0/cargo_home/bitbake/mbox-0.3.1/src/mbox.rs:188:16
    |
188 |         MALLOC <- value.clone()
    |                ^^ expected one of 8 possible tokens here

error: aborting due to previous error

error: Could not compile `mbox`.

Caused by:
  process didn't exit successfully: `rustc --crate-name mbox /yocto/work/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/healthdog/0.1.0.AUTOINC+a10155844f-r0/cargo_home/bitbake/mbox-0.3.1/src/lib.rs --color never --crate-type lib --emit=dep-info,metadata,link -C opt-level=3 --cfg 'feature="default"' --cfg 'feature="std"' -C metadata=5904261533b651af -C extra-filename=-5904261533b651af --out-dir /yocto/work/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/healthdog/0.1.0.AUTOINC+a10155844f-r0/git/target/arm-poky-linux-gnueabi/release/deps --target arm-poky-linux-gnueabi -C linker=/yocto/work/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/healthdog/0.1.0.AUTOINC+a10155844f-r0/wrapper/target-rust-ccld -L dependency=/yocto/work/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/healthdog/0.1.0.AUTOINC+a10155844f-r0/git/target/arm-poky-linux-gnueabi/release/deps -L dependency=/yocto/work/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/healthdog/0.1.0.AUTOINC+a10155844f-r0/git/target/release/deps --extern libc=/yocto/work/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/healthdog/0.1.0.AUTOINC+a10155844f-r0/git/target/arm-poky-linux-gnueabi/release/deps/liblibc-cc88c750fa2562e8.rlib --cap-lints allow -L /yocto/work/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/healthdog/0.1.0.AUTOINC+a10155844f-r0/recipe-sysroot/usr/lib/rust --remap-path-prefix=/yocto/work/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/healthdog/0.1.0.AUTOINC+a10155844f-r0=/usr/src/debug/healthdog/0.1.0.AUTOINC+a10155844f-r0 -C prefer-dynamic -C 'link-args=-Wl,-rpath=$ORIGIN/../lib/rust:$ORIGIN/../lib/arm-poky-linux-gnueabi/rustlib/arm-poky-linux-gnueabi/lib' --cfg stable_channel` (exit code: 1)
WARNING: exit code 101 from a shell command.
ERROR: Function failed: do_compile (log file is located at /yocto/work/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/healthdog/0.1.0.AUTOINC+a10155844f-r0/temp/log.do_compile.427504)

```

Maybe #1723 will solve the problem?

ZubairLK, 2019-11-13:
@kbumsik #1723 is the fix for this indeed.

kbumsik, 2019-11-13:
Great, thanks for letting me know. Since you put `prio/high` tag I'm not sure to close this issue now. Anyway please close this issue if you are not using it as a tracker.

ZubairLK, 2019-11-13:
@kbumsik ah. correct. closing as this is already tracked in #1719

closed 2019-11-15 by kbumsik