balena-os/meta-balena #1747: NetworkManager temp timestamps file may fill the state partition

majorz, 2019-11-15:
Temporary timestamps files are left occasionally on devices which have some unknown filesystem issue. We have not been able to reproduce locally although we tried different approaches.

Similarly to how we clean up journal files we need to delete temporary timestamps files on boot. Those are located in `/mnt/state/root-overlay/var/lib/NetworkManager` and are named `timestamps.XXXXXX`, where XXXXXX is some random string.
Related to: https://bugs.launchpad.net/ubuntu/+source/network-manager/+bug/913856

majorz, 2019-11-15:
Additionally we need to clean any `seen-bssids.XXXXXX` files as well. This is a six symbols extension.

ZubairLK, 2019-11-19:
Things to look into
- are those files even needed? (can this be a folder in tmpfs in ram with a periodic cleanup? systemd tmp files?)
- How NM behaves if that folder(or files in that folder) are removed?

alexgg, 2019-11-20:
I have been looking at the history of this:

- Initially /var/lib was a tmpfs (https://github.com/balena-os/meta-balena/issues/406), and obviously this was affecting packages that needed persistent storage
- This ended up moving the networkmanager state directory to persistent storage (https://github.com/balena-os/meta-balena/pull/1027)

There is a bunch of state database file that NM stores here:

- **timestamps**, which records an active connection last time usage. These could live on non-persistent storage and NM will re-connect to all active connections on boot
- **seen-bssids**, which caches the wireless AP SSIDs. These could also live on non-persistent storage and NM will re-create the cache on boot
- **lease** files, which keeps track of DHCP leases. Again, these could leave on persistent storage as typically DHCP servers will be setup to serve the same IP lease for the same MAC
- **secret_key**, this defines a host ID and needs to be persistent so things don't break

Also, this has been reported on NM v1.14.4 and current balenaOS is at v1.20.2. Things have changed in this area between versions, including the way the files are stored.

https://github.com/NetworkManager/NetworkManager/commit/8a78493de1c33d879082e55edca1ee6e672efc40

Which means these files are only read and written when NM starts/stops.

There are two options as I see it:

1. A brute force boot script that attempts to clean up the stale files. This can easily get out-of-sync with NM and when broken it's likely it will go unnoticed until someone hits the problem again.
2. A patch for NM that moves leases, timestamps and seen-bssids to a tmpfs (probably /run), while leaving the secrets_key in persistent storage.

The second is my preferred option as:

- BalenaOS already controls NM version and maintains a bunch of patches
- The patch can be attempted to be upstreamed so it no longer has to be maintained.

ZubairLK, 2019-11-25:
From https://github.com/NetworkManager/NetworkManager/commit/8a78493de1c33d879082e55edca1ee6e672efc40
```
- our state files tend to grow as we don't garbage collect old entries.
  Keeping this all in memory might be problematic. However, the right
  solution for this is that we come up with some form of garbage
  collection so that the state files are reaonsably small to begin with.
```

That right there is the problem..
Cleaning up just on boot won't make the problem completely go away..
option 2 seems sensible and correct. I'm a bit wary of how invasive the changes might be.

if its a few lines, it should be ok to carry the patch

majorz, 2019-11-25:
The temp files are being created with the atomic write pattern where a temp file is created, content is written and then the temp file is moved over the original file. If the atomic write fails somewhere in the middle a temp file will be left and the original file will not be touched.

This possibly could be because of a faulty SD card or some power issue. Thus it does not seem right to patch NetworkManager to workaround a more general system issue.

In a perfect (future) scenario the user should be notified that something like this is going on with his device and investigative actions could be taken. If we patch NetworkManager we will only obscure the real problem.

ZubairLK, 2019-12-02:
Hmm. Actually a generic partition almost full warning with some dashboard element sounds useful for both state and data partition.

alexgg, 2019-12-02:
@majorz I would argue that if this is caused by a faulty  hardware there is nothing to fix in the software, the hardware needs to be RMAd, replaced, even if it's as simple as swapping SD cards.
By power issue, do you mean losing power in the middle of the atomic move? I'd argue that Linux filesystem wouldn't have synched the temporary file, unless the filesystem is mounted to always syncon write which is not a good idea.

The hypothesis I was working on goes back to what the NM commit referenced above says:
https://github.com/balena-os/meta-balena/issues/1747#issuecomment-558089582

"our state files tend to grow as we don't garbage collect old entries"

So for a long running system, the state files might grow to be a problem. Moving these into memory would workaround a problem as:
1) There is much more memory that space in /mnt/state
2) The files are cleared on reboot

However, if we are not sure what is that we are trying to fix I would just let it be.

@ZubairLK suggestion of a partition full warning, maybe with a manual cleansing, might be the best suggestion.

majorz, 2019-12-02:
I have seen at least three cases of this issue up till now - those are cases that we know of and quite possibly there are ones that we do not. The temporary files have accumulated slowly over a month or more judging from their last modified date. It is not one or two of those temp files left, but lots of them, so the issue is reoccurring continuously.

It is important to note that we never experienced issues with NM state files being anywhere close to filling up the partition in normal circumstances as they are pretty small. The state partition fills in when there are many-many duplicates - temp leftovers.

I have a feeling we are starting to go in circles here and this is a bit worrying :)

commit 6f83fea6 (2019-12-06): networkmanager: Remove stale temporary files on startup

Reports indicate that NetworkManager can leave stale temporary files on
the state partition that over time can affect the device's operability.

This commit removes the timestamps.XXXXXX and seen-bssids.XXXXXX files on
startup to avoid this situation.

Fixes #1747

Change-type: patch
Changelog-entry: Remove networkmanager stale temporary files at startup
commit 64baa24e (2019-12-12): networkmanager: Remove stale temporary files on startup

Reports indicate that NetworkManager can leave stale temporary files on
the state partition that over time can affect the device's operability.

This commit removes the timestamps.XXXXXX and seen-bssids.XXXXXX files on
startup to avoid this situation.

Fixes #1747

Change-type: patch
Changelog-entry: Remove networkmanager stale temporary files at startup
commit c1ea36a5 (2019-12-16): networkmanager: Remove stale temporary files on startup

Reports indicate that NetworkManager can leave stale temporary files on
the state partition that over time can affect the device's operability.

This commit removes the timestamps.XXXXXX and seen-bssids.XXXXXX files on
startup to avoid this situation.

Fixes #1747

Change-type: patch
Changelog-entry: Remove networkmanager stale temporary files at startup
commit 1234b29d (2019-12-23): networkmanager: Remove stale temporary files on startup

Reports indicate that NetworkManager can leave stale temporary files on
the state partition that over time can affect the device's operability.

This commit removes the timestamps.XXXXXX and seen-bssids.XXXXXX files on
startup to avoid this situation.

Fixes #1747

Change-type: patch
Changelog-entry: Remove networkmanager stale temporary files at startup
closed 2019-12-16 by alexgg