balena-os/meta-balena #1748: Fix race condition where /var/run/balena-engine.sock can become a folder

ZubairLK, 2019-11-20:
After fixing https://github.com/balena-os/meta-balena/issues/1650 via PR https://github.com/balena-os/meta-balena/pull/1690

We introduced a subtle race condition. During boot, supervisor container may not find the balenaEngine socket and as a result make `/var/run/balena-engine.sock` into a folder because of the way `-v` works in the balena run command.

The upstream issue is here

https://github.com/moby/moby/issues/30348

A sensible workaround suggested there is to bind mount `/var/run` inside the container instead of the exact path `/var/run/balena-engine.sock`

It is possible to have a one-liner fix for existing devices.

something like

```
mount -o remount,rw / && \
make edit via sed
mount -o remount,ro /
```

roman-mazur, 2019-11-20:
@ZubairLK 
How was this prevented before we changed the health checks? 

Also, is it the same problem we saw in user app containers? So it's not only about the supervisor, but any container that starts with the engine socket bind mount.

I think the difficulty in fixing it lies in the fact that bind mount is a generic thing. But for user app containers we expose the engine socket with a dedicated instruction (a special label on the container), abstracting the bind mount.
If we moved this logic from the supervisor to the engine, the engine could clearly see the container dependency and organize the order of containers start, making sure that the socket exists before starting containers that depend on it (and allowing others to start earlier).

ZubairLK, 2019-11-20:
internal discussion link
https://www.flowdock.com/app/rulemotion/r-resinos/threads/cs_lLeY9v52H-AuH7hUobpBZMop

ZubairLK, 2019-11-20:
First pass at a scripted fix. Untested.

```
mount -o remount,rw / && \
mv /usr/bin/start-resin-supervisor /usr/bin/start-resin-supervisor.new && \
sed -i 's#/var/run/balena-engine.sock#/var/run#g' /usr/bin/start-resin-supervisor.new && \
mv /usr/bin/start-resin-supervisor.new /usr/bin/start-resin-supervisor && \
mount -o remount,ro / || true

# If we have failed already, recover
if [ -d /var/run/balena-engine.sock ]; then
        rm -rf /var/run/balena-engine.sock
        systemctl daemon-reload
        systemctl stop resin-supervisor
        systemctl restart balena-engine.socket
        systemctl restart balena
        systemctl start resin-supervisor
fi
```

closed 2019-11-21 by ZubairLK

ZubairLK, 2019-12-04:
The fix is https://github.com/balena-os/meta-balena/commit/7cf43d9f939b1d960cb677c48ca9f91b668d2dc3

commit 1f28dc42 (2019-11-20): resin-supervisor: Fix for race condition when supervisor starts up

Fixes #1748

Change-type: patch
Changelog-entry: Fix for a race condition where occasionally the supervisor might not be able to come up during boot. Also can be caused by using io.balena.features.balena-socket
and app container restart always policy. Affects meta-balena 2.44.0 and 2.45.0
commit 0c74a2f2 (2019-11-20): resin-supervisor: Fix for race condition when supervisor starts up

Fixes #1748

Change-type: patch
Changelog-entry: Fix for a race condition where occasionally the supervisor might not be able to come up during boot. Also can be caused by using io.balena.features.balena-socket
and app container restart always policy. Affects meta-balena 2.44.0 and 2.45.0
commit 0fbac6f0 (2019-11-21): resin-supervisor: Fix for race condition when supervisor starts up

Fixes #1748

Change-type: patch
Changelog-entry: Fix for a race condition where occasionally the supervisor might not be able to come up during boot. Also can be caused by using io.balena.features.balena-socket and app container restart always policy. Affects meta-balena 2.44.0 and 2.45.0. To be fixed in 2.44.1 and 2.46.0
commit 7cf43d9f (2019-11-21): resin-supervisor: Fix for race condition when supervisor starts up

Fixes #1748

Change-type: patch
Changelog-entry: Fix for a race condition where occasionally the supervisor might not be able to come up during boot. Also can be caused by using io.balena.features.balena-socket and app container restart always policy. Affects meta-balena 2.44.0 and 2.45.0. To be fixed in 2.44.1 and 2.46.0