balena-os/meta-balena #1756: Measure bandwidth consumption on every release

ZubairLK, 2019-11-22:
There can be many ways. Real ways with expensive hardware equipment/router measuring etc.

Or there can be a project/app container we push to the device that is privileged and measures on-device band-width utilisation. It'll be good enough to get a rough metric and it will be easy to spot if something crazy happens between releases.

Should test in two configurations:-
1 - Default prod image. 1 hour running that container
2 - Default prod image. 1 hour running that container but with everything switched off. i.e. logging/vpn disabled etc.

Device. Probably a fin until we have a reference platform in meta-balena

alexgg, 2019-11-26:
I have an initial test on a RPI4 on:
```
ID="balena-os"
NAME="balenaOS"
VERSION="2.44.0+rev3"
VERSION_ID="2.44.0+rev3"
PRETTY_NAME="balenaOS 2.44.0+rev3"
MACHINE="raspberrypi4-64"
VARIANT="Development"
VARIANT_ID="dev"
META_BALENA_VERSION="2.44.0"
RESIN_BOARD_REV="a4f4d1a"
META_RESIN_REV="7fed82f"
SLUG="raspberrypi4-64"
```
The device is configured as follows:
[image]

- No VPN connectivity check
- No logs to Balena
- DIsabled VPN
- No persistent logging

Running the app at:
https://github.com/alexgg/docker-app-bwc
I have left bmon and tcpdump running for 10 minutes.
[image]
With the wireless unconfigured there was only traffic on eth0:
```RX bytes: 11.63 MiB
TX bytes: 5.10 MiB```
Analysing the tcpdump:
```# Hosts
#
# 6 entries.

192.168.1.118	b06608a.local
34.237.229.125	resin-vpnnl-s1wro58y28co-67b66e9f6dbf0e29.elb.us-east-1.amazona
3.215.161.68	resin-d-alb-15sj6u7hs9afn-1069262607.us-east-1.elb.amazonaws.co
18.232.192.190	resin-vpnnl-s1wro58y28co-67b66e9f6dbf0e29.elb.us-east-1.amazona
34.200.210.179	resin-d-alb-15sj6u7hs9afn-1069262607.us-east-1.elb.amazonaws.co
fe80::6f19:83c7:d196:fe98	b06608a.local```

It all seems to be API traffic from the supervisor to BalenaCloud.

I will build on this running for 1h and also trying on other released versions.

alexgg, 2019-11-28:
Improving on the above, I have now changed the poll interval to 24 hours and that significantly improves the numbers.

Over a 9h long run, the results are:

**RPI4 @ 2.41.0+rev4**
RX: 28 MB, 3.11 MB/h
TX: 11.08 MB, 1.23 MB/h

Over an 8 hours run:

**BalenaFin @ 2.38.0+rev1**
RX: 15.30 MB, 1.91 MB/h
TX: 4.08 MB, 0.51 MB/h

I still need to test on 2.39.0+rev1, and I would like to repeat with the BalenaFin in the same revision as the RPI4 to confirm the numbers are similar.

Anyone, this is far from the 1.3MB per month measured previously.

ZubairLK, 2019-11-28:
Another test to isolate the increase in bw would be to use fin 2.45 but remove new supervisor and run the old 2.38 supervisor. That way we'll know if something in OS changed. or the likely suspect which is the supervisor at this point.

I think there is once an hour connectivity checking via networkmanager introduced in 2.34. I wonder how much impact that has. afaik, we don't have a supervisor/dashboard way of changing those values. Its via config.json
https://github.com/balena-os/meta-balena#connectivity

I sincerely hope a connectivity check once an hour is low overhead..

alexgg, 2022-04-06:
Note that these measurements were taken using an IPv4 network and IPv6 increases these numbers significantly with the use of ICMP.