balena-os/meta-balena #1762: Profile memory usage differences between 2.31 and 2.44

xginn8, 2019-11-25:
Previously functional apps sometimes now die due to memory constraints, so there may be memory bloat between these two OS versions.

There is a workaround to save some memory by killing `plymouthd` at runtime:

```
DBUS_SYSTEM_BUS_ADDRESS=unix:path=/host/run/dbus/system_bus_socket dbus-send --system --print-reply --dest=org.freedesktop.systemd1 /org/freedesktop/systemd1 org.freedesktop.systemd1.Manager.KillUnit string:"plymouth-start.service" string:"all" int32:"15"
```

floion, 2020-05-28:
On a pi3 A+, running 2.47.0+rev1:

>  814     1 root     S     970m 203%   0% /usr/bin/balenad --experimental --log-driver=journald -s aufs -H fd:// -H unix:///var/run/balena.sock -H unix:///var/run/balena-engine.sock -H tcp://0.0.0.0:2375 --dns 10.114.102.1 --bip 10.114.1
  860   814 root     S     938m 196%   0% balena-engine-containerd --config /var/run/balena-engine/containerd/containerd.toml --log-level info
 1691   860 root     S     890m 186%   0% balena-engine-containerd-shim -namespace moby -workdir /var/lib/docker/containerd/daemon/io.containerd.runtime.v1.linux/moby/b65dfac7ca55ca088ffbbab4c82ade112940431407fce0709f47f8665b23952c -addr
 1107   860 root     S     890m 186%   0% balena-engine-containerd-shim -namespace moby -workdir /var/lib/docker/containerd/daemon/io.containerd.runtime.v1.linux/moby/91ce86c0c725f7538325e4ef198698cd5906c6fd97bbb024dcf519f3f656198d -addr
 1682  1601 root     S     882m 184%   0% balena start --attach resin_supervisor
 1710  1691 root     S     149m  31%   0% node /usr/src/app/dist/app.js

we have:

> root@c6fe132:~# cat /proc/1107/status | grep -i vmrss
VmRSS:       29256 kB
root@c6fe132:~# cat /proc/814/status | grep -i vmrss
VmRSS:       40736 kB
root@c6fe132:~# cat /proc/860/status | grep -i vmrss
VmRSS:       33604 kB
root@c6fe132:~# cat /proc/1691/status | grep -i vmrss
VmRSS:       29288 kB
root@c6fe132:~# cat /proc/1710/status | grep -i vmrss
VmRSS:     65176 kB

robertgzr, 2020-05-29:
@floion note that `VmRss` is the size of the (loaded) binary+libs *and* the actual heap/stack and between the daemon, containerd, the shims and the balena-engine cli they all share the same binary+libs, which should be cached only once.

alexgg, 2020-05-29:
@floion @robertgzr A different approach to overcome the problem with `vmRss` including shared memory pages is to take a snapshot of the total free memory. Doing that for with a RPI4 1GB running 2.50.0 I got the following numbers.

Use cases:

- baseline:  The OS inc. balena-engine, no resin-supervisor
- resin-supervisor without apps: No running apps
- one idle Alpine app: One simple Alpine app on sleep loop
- two idle Alpine apps:  Two simple Alpine apps on a sleep loop

Numbers, used memory in MiB:

- baseline: ~188MiB used
- resin-supervisor without apps:  ~286M used (delta of 98M)
- one idle Alpine app: ~340M (delta of 55M, 52MB used by the supervisor)
- two idle Alpine apps:  ~364M (delta of 24M, 54M by the supervisor, 3MB both apps)

@floion @robertgzr it would be great if the numbers above could be peer reviewed