balena-os/meta-balena #1765: Ensure GSM connectivity

simplelife2010, 2019-11-30:
This is a feature request related to
https://forums.balena.io/t/help-with-debugging-gsm-needed/47808

I was told it is common that modem connections are lost and not recovered automatically by NetworkManager/ModemManager.

It is crucial that remotely installed devices stay online reliably by all means. The suggestion was to check the network connectivity and reset the modem when necessary. I think this should be implemented as a balenaOS feature, because it is so important and should be working without any containers deployed.

It should not need to be implemented by a balena user.

ZubairLK, 2019-12-02:
@simplelife2010 

This has come up before. Unfortunately, there is no easy fix on the OS side. 

> and should be working without any containers deployed

Are you using balenaOS without any containers by any chance?

balenaOS tries to keep all services running, balenaEngine, supervisor, NetworkManager, ModemManager, vpn etc. Now in this case, ModemManager itself is fine, but the actual internet connectivity isn't and a restart (of connection/service/kernel module/firmware load/reload/reboot?) may or may not fix it.

Internet connectivity checking isn't trivial as it is bandwidth intensive (something we want to conserve when using modems.) Also, people have different scenarios/requirements w.r.t. firewalls as well.

Hence, up until now, this has been left to the application to figure this out and restart the modem if needed.

In an ideal situation, we could provide a small container that only does that i.e. keeps modem alive and has internet connectivity checking. And you could include that container in your application.

simplelife2010, 2019-12-04:
I understand. I would be interested in such a watchdog container.

seatosummit, 2019-12-04:
I too would be interested in such a container. This would alleviate some
headaches for us.

aessig, 2021-07-27:
@ZubairLK any idea when or how this could happen? Thanks 

> In an ideal situation, we could provide a small container that only does that i.e. keeps modem alive and has internet connectivity checking. And you could include that container in your application.

karandwivedi42, 2023-02-06:
> Hence, up until now, this has been left to the application to figure this out and restart the modem if needed.

Can someone please point to any guidance on how we can implement it on our own?