balena-os/meta-balena #1774: Enable fstrim.timer by default

Ereski, 2019-12-10:
`fstrim.timer` runs periodically runs fstrim from util-linux which sends a TRIM command to supported SSDs. TRIM is helpful to prevent lifetime and performance degradation, specially on older and cheaper SSDs.

alexgg, 2019-12-10:
@floion probably it's a good idea to do this for any device type that has a sata interface. I am thinking about whether adding a new "sata" machine feature that individual devices could define. Comments?

petrosagg, 2019-12-10:
Why not just mount the filesystem with the `discard` option?

Ereski, 2019-12-11:
That enables continuous TRIM which doesn't seem to be the recommended option due to performance issues: https://wiki.archlinux.org/index.php/Solid_state_drive#Continuous_TRIM.

alexgg, 2022-01-21:
Also, it might be better to wait for the 5.18 kernel, see https://github.com/raspberrypi/linux/pull/4832#issuecomment-1017900603