balena-os/meta-balena #1776: `time-sync.target` is improperly reached with chronyd

xginn8, 2019-12-17:
`chronyd` is a `Type=simple`, and therefore considered active as soon as the process forks. Since it is ordered before `time-sync.target`, time is assumed to be synchronized once the `chronyd` process is forked. Follow-on processes that actually require time to be _fully_ synchronized are started immediately, while `chronyd` is still forcing clock changes.

Specifically, this results in a connection cycle (`ping-restart`) for `openvpn`, since the clock changes and `openvpn` thinks it hasn't received a ping in a time that's over the threshold. cc @wrboyce @alexgg

ZubairLK, 2019-12-31:
@xginn8 in an ideal world, we could wait for real time sync before reaching `time-sync.target`.
However, that would prohibit other services from running until NTP sync is reached. NTP sync may or may not happen depending on connectivity.

Time does jump in the OS. See https://github.com/balena-os/meta-balena#time-in-the-os
If chrony is making large jumps (which it does initially), while openvpn is trying to connect, I'm afraid openvpn will have to be able to handle that.

I don't think the solution to #1779 is in time-sync but open to ideas and suggestions

xginn8, 2019-12-31:
@ZubairLK, imho we should order units that _strictly_ require time synchronization after `time-sync.target` (using something like [`ntp-wait`](https://linux.die.net/man/8/ntp-wait)?). as is, since `time-sync.target` merely indicates that `chrony` has forked, then we should order units that want time sync as such using `chrony`. imho it provides a false sense of security about the state of time on the device, and makes debugging harder as we do not abide by the spirit of `time-sync.target` from https://www.freedesktop.org/software/systemd/man/systemd.special.html#time-sync.target.

closed 2021-12-01 by bulldozer-balena[bot]