balena-os/meta-balena #1805: Update ModemManager to v1.12.4 (latest stable) and corresponding libraries

majorz, 2020-01-15:


ZubairLK, 2020-01-17:
@majorz is there any concrete reason for going for this (other than bumping to latest stable?)

The reason I ask is that I'd like to slowly avoid moving ahead of Yocto and align with Yocto recipes for NM/MM.

Yocto is at 1.10.8
https://github.com/openembedded/meta-openembedded/tree/master/meta-oe/recipes-connectivity/modemmanager

There are some benefits to keeping in line with Yocto.

majorz, 2020-01-17:
@ZubairLK - yes, I created it by request of @shaunmulligan here: https://www.flowdock.com/app/rulemotion/r-resinos/threads/U-D2rB_kXCPz-0QhhAMyOsSE39t

shaunmulligan, 2020-01-17:
@ZubairLK i suggested it since there has been a fair bit of work on MM with their maintainer and some of our users to fix specific issues around memory leaks, etc that I think we would want to bring in. I think sticking to  yocto versions is a bad idea. balenaOS should prescribe versions of the specific core components that we rely on, so that people know that it won't change under them just because a BSP bumped yocto version

floion, 2020-01-20:
Created https://github.com/balena-os/meta-balena/pull/1810

commit c7ab22dc (2020-01-29): modemmanager: Update to v1.12.4

Fixes #1805

Change-type: minor
Changelog-entry: Update to ModemManager v1.12.4
commit 3096e40b (2020-01-29): modemmanager: Update to v1.12.4

Fixes #1805

Change-type: minor
Changelog-entry: Update to ModemManager v1.12.4
commit fd45ee88 (2020-01-29): modemmanager: Update to v1.12.4

Fixes #1805

Change-type: minor
Changelog-entry: Update to ModemManager v1.12.4
closed 2020-01-29 by ZubairLK