balena-os/meta-balena #1812: NetworkManager occasionally does not reconnect to WiFi networks

majorz, 2020-01-20:
This does not happen often, but we have a user who is more severely affected by it. Rebooting makes the device reconnect again.

For now I implemented the following workaround that we are still verifying whether it resolves the issue or not (WIP): https://github.com/balena-io-playground/keep-wifi-up

I reproduced the issue once locally, but I has been focused on the workaround, so I still have to find a way to reproduce in a regular manner.

We had other instances reported and at some point other team members were also able to reproduce this occasionally.

majorz, 2020-02-03:
Implemented a workaround here while we get to the bottom of this: https://github.com/balena-io-playground/keep-wifi-up

Hades32, 2021-06-09:
Discussed also in https://www.flowdock.com/app/rulemotion/r-resinos/threads/d50POtyAhzXDqA-fUk-OQVhldW-

Reproduced on a NUC and a notebook hotspot:
[wpa.log](https://github.com/balena-os/meta-balena/files/6626458/wpa.log)
[nm.log](https://github.com/balena-os/meta-balena/files/6626460/nm.log)

Hades32, 2021-06-10:
Retested with mac randomization turned off. Same behavior: In the time-frame the logs are showing, I was connected with Ethernet, then I enabled the hotspot, then the device connected to the WiFi hotspot, then I disabled the hotspot and enabled it again ~10s later. The device never reconnected...

[wpa.log](https://github.com/balena-os/meta-balena/files/6629224/wpa.log)
[nm.log](https://github.com/balena-os/meta-balena/files/6629226/nm.log)