balena-os/meta-balena #1815: Enable IRQ sharing for 8250 serial driver

alexgg, 2020-01-22:
Enable serial IRQ sharing for all devices. This can be tested on the Intel NUC as reportedly only 2 of the 6 available serial ports are working without it.
Should not affect platforms that don't share IRQs, but will require a sanity test across a selection of boards.

Basically add SERIAL_8250_EXTENDED and SERIAL_8250_SHARE_IRQ.

erikhh, 2020-01-22:
I'm no Linux kernel expert at all, so this is just a suggestion. Would it be possible to compile the 8250 driver as a kernel module? 
That way you can change the driver to your liking without having to deal with a full Yocto build.

alexgg, 2020-01-23:
Hi @erikhh, it's possible, but usually the serial driver is built-in so that you can use the serial port when debugging even if there is no user space to load modules.
Also, by changing the driver to you liking you mean cross-compiling your own module that matches the running kernel and installing it I guess. The problem with this approach is that the changes would be lost each time the host OS is updated, so in the end the module in balenaOS would also need to be compiled with the desired configuration.