balena-os/meta-balena #1829: OS should try and fetch the supervisor version it needs to run in case of a failed super update during HUP

floion, 2020-02-03:
Tried in staging a hup from 2.41.0+rev4 to 2.47.0+rev1 on a rpi4.
However, seems like there were some API hiccups right at the moment of the update:

> [000000050][LOG]Supervisor update start...
[000000050][LOG]No explicit supervisor version was provided, update to default version in target resinOS...
[000000052][LOG]Extracted default version is v10.6.27...
parse error: Invalid numeric literal at EOF at line 1, column 12
[000000053][WARNING]Could not parse current supervisor version from the API, skipping update...
[000000054][LOG]No update package cleanup done
[000000054][LOG]Rebooting into new OS in 5 seconds...

This made the host OS update the OS only, not the supervisor also. Upon reboot, the new OS 2.47.0+rev1 was running the old supervisor version 10.2.2 instead of the new 10.6.27. And also:

> # cat /etc/resin-supervisor/supervisor.conf 
# This file represents the last known version of the supervisor
SUPERVISOR_IMAGE=balena/aarch64-supervisor
SUPERVISOR_TAG=v10.2.2
LED_FILE=/sys/class/leds/led0/brightness

My opinion is that the new OS should instead have the supervisor.conf pointing to the new supervisor version and try to update to that one.

floion, 2020-02-03:
Run the supervisor update manually and got this:

> Extracted supervisor ID: 6175
Supervisor configuration found from API.
Error: No such object: balena/aarch64-supervisor:v10.6.27

So there may be connectivity issues when fetching the supervisor update. Which further makes the point for an OS functionality that should work for periodically fetching the correct supervisor version that needs to be running in that particular OS version.

floion, 2020-02-03:
I also got this other logs from another failure:

> VL805 FW version: 000137ab
[INFO] vl805 firmware update is not necessary
[INFO] Switching uboot root partition index to 3...... done.
New hooks ran successfully.
[000000060][LOG]Supervisor update start...
[000000060][LOG]No explicit supervisor version was provided, update to default version in target resinOS...
[000000062][LOG]Extracted default version is v10.6.27...
[000000063][LOG]Supervisor update: will be upgrading from v10.2.2 to v10.6.27
[000000063][LOG]Extracted supervisor vars: ID: 6175
[000000063][LOG]Setting supervisor version in the API...
[000000064][LOG]Running supervisor updater...
[000000090][LOG]Stopping supervisor and related services...
parse error: Invalid numeric literal at EOF at line 1, column 12

floion, 2020-02-03:
Related: https://github.com/balena-io/balena-supervisor/issues/1189

alexgg, 2020-02-06:
Not sure about whether this is the right thing to do. It seems to me that the way the system is going with multi-app is to decouple even more the supervisor and the OS, and even split the supervisor into several apps for specific features.
Unless this is a critical issue affecting customers, I'd probably delay taking any decision at the OS level until the multi-app architecture develops.
CC: @floion @ZubairLK @CameronDiver @petrosagg