balena-os/meta-balena #1841: Updating the supervisor version should always honour the command args

richbayliss, 2020-02-25:
https://github.com/balena-os/meta-balena/blob/a7c5f47ef43c9ef79b770ec6ac83989130f5f4ed/meta-balena-common/recipes-containers/resin-supervisor/resin-supervisor/update-resin-supervisor#L93-L99

I think this predicate should be reversed; we currently ignore commandline args in favour of the API-held versions. If a user updates the supervisor via the dashboard then the target version is pinned in the API and then later on a shell session cannot update to the version passed in manually.

ZubairLK, 2020-02-25:
So an aspect of `update-resin-supervisor` was discussed in Todays arch call (25th Feb 2020. First item)

The state set in the API is supposed to be the device state. By default, the state is empty and hence no action is taken. But once the state is set in the API backend, the device really should be on that state.

If we reverse this ordering, there are other things to consider. e.g. There is a timer in the OS. `update-resin-supervisor.timer` that will update the supervisor to the version configured in the API.

But now we would like to override the API state via the command line args.

One way this feature request can be accommodated I guess would be to add a `--force/--clear-api` flag to `update-resin-supervisor`

That would clear the API of any state and will be an active opt-in on part of the user. This will be then useful for debugging mode which I suspect is what Rich wants to use?

And combined with the `-t` flag, should then work

Does that sound sensible @richbayliss @CameronDiver ?

CameronDiver, 2020-02-26:
@ZubairLK clearing the API field makes me worried. One of the main things we've spoken about in the arch call is that we should never ever downgrade the supervisor. To this end we should avoid ever clearing the field so we have something to compare against when an upgrade (or downgrade) request comes in.

alexgg, 2022-01-19:
As discussed above, the supervisor version should be dictated by either the preloaded version, or the API. No manual override should be possible.
Currently the supervisor is no longer retrieved from dockerhub but from the internal registry, so it can no longer be pulled from a combination of repository:version. In this context, manually passing repository version does not make sense so we will remove the option of manual override.

closed 2022-01-19 by alexgg