balena-os/meta-balena #1846: NetworkManager times out on pppd when setting up UART modems

majorz, 2020-03-09:
Reproduces with Sixfab Raspberry Pi 3G/4G-LTE Base HAT when running over UART. After ModemManager initializes the modem, NetworkManager launches pppd successfully, but afterwards the pppd plugin does not respond back through D-Bus: `pppd timed out or didn't initialize our dbus module`. There are reports of other modem models with the same issue, so it is not limited to the model we test with. I am currently debugging this.

roman-mazur, 2020-03-09:
Extra info from the original reporter:

>it looks like adding an empty connection string in the ppp settings serves as a workaround.
>So if the `/etc/ppp/options` file has a connect "", it doesn't fail

TSKnuhtsen, 2020-04-23:
I will be a paying customer when this is fixed

GitTibbe, 2020-06-16:
Any update on this issue? @majorz

TSKnuhtsen, 2020-06-25:
Can't use a proper case on my RaspberryPi4B without this fixed :(

qvntra, 2020-06-29:
@majorz has there been headway made on this feature?

majorz, 2020-07-06:
This is still being worked on. I got a couple of incidents with this issue as my first test modem broke, and the next one I ordered could use pppd through UART at all, so the investigation was delayed as I needed to switch on working on a other tasks. I will be resuming this as soon as possible though.

Can you please list the modem models you are facing this issue with?

GitTibbe, 2020-07-06:
> This is still being worked on. I got a couple of incidents with this issue as my first test modem broke, and the next one I ordered could use pppd through UART at all, so the investigation was [...]

I used this device https://doc.rakwireless.com/datasheet/rakproducts/rak7243c-pilot-gateway-datasheet which is basically a raspberry pi 3 + QUECTEL EG95 connected via a HAT (UART). (I had to add the correct UDEV rules on ttsy0)

majorz, 2020-07-06:
@GitTibbe Is using USB connector an option for you? That is connecting the USB cellular port on the case to one of the RPi USB ports with a cable. This will make it use the qmi driver and will give you a higher data rate. Or you would like to avoid having this external cable altogether?

GitTibbe, 2020-07-06:
> @GitTibbe Is using USB connector an option for you? That is connecting the USB cellular port on the case to one of the RPi USB ports with a cable. This will make it use the qmi driver and will give you a higher data rate. Or you would like to avoid having this external cable altogether?

@majorz unfortunately that's not an option for production as the product comes in an enclosure and it will look strange for customers to have a cable poking out and in. 

But we know that adding connect "" to "/etc/ppp/options" works, which should be an easy to fix in BalenaOS right?

majorz, 2020-07-06:
> But we know that adding connect "" to "/etc/ppp/options" works, which should be an easy to fix in BalenaOS right?

Yes, this would be simple to apply. We still need to get to the bottom of why this option change is needed and why the default one does not work. Figure out the proper way to fix it without obscuring other issues. The other team members were not okay with applying such change without going through all of that.

qvntra, 2020-07-09:
@majorz Quectel EC25 Mini PCle 4G/LTE Module on RaspberryPi4B

commit e8940142 (2021-01-12): ppp: Fix timeout when launched by NetworkManager

Fixes #1846

UART modems (RaspberryPi HATs) are not working currently under
balenaOS as NetworkManager times out while attempting to establish
ppp connection. This is not a balenaOS specific issue.

This commits adds a `connect ""` declaration to `/etc/ppp/options`
to workaround this as the NULL default value causes the timeout.

The connect option specifies an external script to establish the
physical link. When using NetworkManager/ModemManager it is
ModemManager that establishes the physical link before passing it
to NetworkManager. Thus `co [... cut]
commit 32deac11 (2021-01-12): ppp: Fix timeout when launched by NetworkManager

Fixes #1846

UART modems (RaspberryPi HATs) are not working currently under
balenaOS as NetworkManager times out while attempting to establish
ppp connection. This is not a balenaOS specific issue.

This commits adds a `connect ""` declaration to `/etc/ppp/options`
to workaround this as the NULL default value causes the timeout.

The connect option specifies an external script to establish the
physical link. When using NetworkManager/ModemManager it is
ModemManager that establishes the physical link before passing it
to NetworkManager. Thus `co [... cut]
commit e86c83e9 (2021-01-14): ppp: Fix timeout when launched by NetworkManager

Fixes #1846

UART modems (RaspberryPi HATs) are not working currently under
balenaOS as NetworkManager times out while attempting to establish
ppp connection. This is not a balenaOS specific issue.

This commits adds a `connect ""` declaration to `/etc/ppp/options`
to workaround this as the NULL default value causes the timeout.

The connect option specifies an external script to establish the
physical link. When using NetworkManager/ModemManager it is
ModemManager that establishes the physical link before passing it
to NetworkManager. Thus `co [... cut]
PR #2093 (2021-01-14): ppp: Fix timeout when launched by NetworkManager
Fixes #1846

UART modems (RaspberryPi HATs) are not working currently under
balenaOS as NetworkManager times out while attempting to establish
ppp connection. This is not a balenaOS specific issue.

This commits adds a `connect ""` declaration to `/etc/ppp/options`
to workaround this as the NULL default value causes the timeout.

The connect option specifies an external script to establish the
physical link. When using NetworkManager/ModemManager it is
ModemManager that establishes the physical link before passing it
to NetworkManager. Thus `con [... cut]
closed 2021-01-14 by bulldozer-balena[bot]