balena-os/meta-balena #1850: Review default power management setting for wireless LAN

alexgg, 2020-03-13:
Power management seems to be turned on in;
https://github.com/balena-os/meta-balena/blob/v2.47.1/meta-balena-common/recipes-support/resin-init/resin-init/resin-init#L7-L10

While the intention seems to be to turn it off:
https://github.com/balena-os/meta-balena/commit/adf69fb8c359a2a34d9f091c3e5270c7f8d246e0

alexgg, 2020-04-17:
Related, https://github.com/balena-os/meta-balena/issues/1422

mtoman, 2020-04-27:
So the script says it disables power save but then it issues
```bash
iw dev wlan0 set power_save on
```
adf69fb actually changes `off` to `on` so unless I am missing something it should be enough to replace it with the (probably) intended
```bash
iw dev wlan0 set power_save off
```

alexgg, 2020-04-27:
hi @mtoman there are several things:

1) The command above applies to the power save feature on devices that use the mac80211 layer. It makes sense to have it off by default but allow applications to turn it on if they want.

So the first thing would be to find a mac80211 wifi card and test whether the on/off have the expected effect. There is information about what to expect on a wireshark capture in https://dot11zen.blogspot.com/2018/02/80211-power-management-with-packet.html

2) The rule above only applies to wlan0, but there is no guarantee that the interface will present itself like that.  We need a udev rule like explained in https://github.com/balena-os/meta-balena/commit/adf69fb8c359a2a34d9f091c3e5270c7f8d246e0

3) We only list the bcm43143 device as supported out-of-the box (see https://www.balena.io/docs/reference/hardware/wifi-dongles/). We need to check whether this is a mac80211 based card or not. If it is, then let's test with that. If not, we may need to find out what's the status for power saving with this specific card.

majorz, 2020-04-30:
I am not sure mac80211 is a requirement for this as any nl80211 driver could implement the `NL80211_CMD_SET_POWER_SAVE` command (including those that do not use mac80211).

mtoman, 2020-05-01:
We have agreed to update `resin-init` to disable power save on `wlan0` to restore the previous state as a short-term solution and deal with the udev approach separately.

commit 6aec8dc1 (2020-05-01): resin-init: turn off wlan0 power save by default

The power save has been switched from 'off' to 'on' in adf69fb
with no apparent motivation, we consider it an oversight
and the desired state is 'off'.

Fixes #1850

Change-type: minor
Changelog-entry: Turn off wlan0 power save
commit 514eec10 (2020-05-05): resin-init: turn off wlan0 power save by default

The power save has been switched from 'off' to 'on' in adf69fb
with no apparent motivation, we consider it an oversight
and the desired state is 'off'.

Fixes #1850

Change-type: minor
Changelog-entry: Turn off wlan0 power save
closed 2020-05-04 by alexgg