balena-os/meta-balena #1852: timedatectl says "System clock is not synchronized" - balenaFin 2.48.0+rev1

floion, 2020-03-29:
Device up for 30 minutes and timedatectl says the following:

> root@67e3989:~# timedatectl 
               Local time: Sun 2020-03-29 05:37:19 UTC
           Universal time: Sun 2020-03-29 05:37:19 UTC
                 RTC time: Sun 2020-03-29 05:37:19
                Time zone: n/a (UTC, +0000)
System clock synchronized: no
              NTP service: n/a
          RTC in local TZ: no

but the date and time printed there are actually correct.

and:

> root@67e3989:~# journalctl --no-pager -u chronyd
-- Logs begin at Fri 2020-01-31 10:38:40 UTC, end at Sun 2020-03-29 05:38:32 UTC. --
Mar 29 05:08:36 localhost chronyd[741]: 2020-03-29T05:08:36Z chronyd version 3.4 starting (+CMDMON +NTP +REFCLOCK +RTC -PRIVDROP -SCFILTER -SIGND +ASYNCDNS -SECHASH +IPV6 -DEBUG)
Mar 29 05:08:55 67e3989 chronyd[741]: 2020-03-29T05:08:55Z Selected source 162.159.200.1

floion, 2020-03-30:
Same thing observed on `skx2` at `v2.48.0+rev1`

floion, 2020-05-25:
Same behavior on Intel NUC at `2.50.1+rev1`

robertgzr, 2020-05-26:
similar behavior on rpi3

robertgzr, 2020-05-26:
```
~# chronyc sources
210 Number of sources = 4
MS Name/IP address         Stratum Poll Reach LastRx Last sample               
===============================================================================
^- static-5-103-128-88.ip.f>     1  14     0   50h   +521us[+1613us] +/- 3316us
^? sweetums.eng.tdc.net          0  14     0     -     +0ns[   +0ns] +/-    0ns
^- 78.156.103.10                 2  14     0   50h    +37us[  +37us] +/- 9878us
^- time.cloudflare.com           3  14     0   50h   +444us[ +444us] +/- 5425us
```

```
~# timedatectl
               Local time: Tue 2020-05-26 13:08:35 UTC
           Universal time: Tue 2020-05-26 13:08:35 UTC
                 RTC time: n/a
                Time zone: n/a (UTC, +0000)
System clock synchronized: no
              NTP service: n/a
          RTC in local TZ: no
```

> From the above I see that the last sync was 50 hours ago, however systemd reports no sync was done. Probably the device was rebooted in the meantime, so chrony has some state and does not need to synchronize, where systemd (timedatectl) thinks it is not synced

from @majorz who was looking into it

xginn8, 2020-06-18:
Seen on 2 RPi3s that had only one configured source until daemon restart:

```
root@1b5cf49:~# chronyc sources
210 Number of sources = 1
MS Name/IP address         Stratum Poll Reach LastRx Last sample               
===============================================================================
^? dc1-recdns02.bellmtsdata>     2  14     3  138m  -362.4s[-362.4s] +/-  113ms
```

```
root@c3c5a49:~# chronyc sources
210 Number of sources = 1
MS Name/IP address         Stratum Poll Reach LastRx Last sample               
===============================================================================
^? 5.186.54.149.static.fibi>     2  14     3  137m  -10897s[-10897s] +/-   40ms
```

markcorbinuk, 2020-06-30:
I'm going to close this issue based on the update to treat default servers as pools. This issue can always be re-opened if problems persist.

floion, 2020-08-04:
@markcorbinuk then the test needs to be changed?

floion, 2020-08-04:
The issue is still present on `2.53.11+rev1` running on Intel NUC with the genericx86-64-ext image.

acostach, 2020-10-23:
Issue present in genericx86-64-ext running on Intel NUC, version 2.58.6+rev1

saintaardvark, 2020-10-28:
See https://github.com/balena-os/balena-intel/issues/292

alexgg, 2020-10-30:
Connects to: https://github.com/balena-os/balena-intel/issues/292

markcorbinuk, 2020-11-03:
The clock synchronisation status as reported by `timedatectl` is determined by checking that the `maxerror` value in the kernel `timex` structure is less than 16000000 (16 seconds). If the value is below this then the time is considered to be synchronised. Upon start-up the `maxerror` field is initialised to 16000000 to indicate that the time is 'unsynchronised'.

Once chrony has received 3 good NTP replies it writes the time to the kernel (via the `adjtimex` call) and sets the `maxerror` field to an appropriate value which in turn indicates to `timedatectl` that the system time is now synchronised. It seems that the value written to `maxerror` by chrony is the 'estimated error' that it displays as the last column of the `chronyc sources` output. In the following example it would write a value of 9776:
```
MS Name/IP address Stratum Poll Reach LastRx Last sample               
=============================================================
^* 162.159.200.123    3     5    377    13    +43us[  +47us] +/- 9776us
```
The linux kernel has a mechanism that automatically increments the `maxerror` field by 500 (usec) once a second, so as long as the kernel time structure is written to by chrony before the `maxerror` value reaches 16000000 time will remain 'synchronised'. In the absence of any chrony updates the 'unsynchronised' state will be reached in 8 hours 53 minutes and 20 seconds. With the current chrony default poll period of 4.5 hours `timedatectl` will report that time is not synchronised after just 2 missed/bad NTP replies.

So, in reality, the status reported by `timedatectl` is completely independent of the status reported by `chronyc sources` other than indicating that the kernel time hasn't been set for approximately 9 hours. Chrony can still show a 'good' current NTP source in this case. Chrony sources are not marked as unreachable until there have been 9 consecutive missed/bad NTP replies (approx. 40 hours) and even when this occurs it doesn't appear that chrony explicitly sets any 'unsynchronised' status. I also can't find any evidence that `systemd-timesyncd` behaves any differently - i.e. it just sets the time (and maxerror) upon receipt of a good NTP reply and lets timedatectl determine the synchronisation status from the `maxerror` value.

Looking at the various user reported issues it appears that there are some misconceptions relating to this behaviour, namely that
'unsynchronised == chrony not running' or 'unsynchronised == system time not set', both of which are not necessarily true.

alexgg, 2020-11-12:
@markcorbinuk it's great to see that level of detail, great digging! Looking at our time synchonization architecture diagram, do you know what does `chronyc waitsync` report after an NTP poll miss? I am thinking that the NTP sync check in the timesync-remote flow should enter a burst loop that will make this not happen.

markcorbinuk, 2020-11-16:
@alexgg It seems that `chrony waitsync` is only useful for detecting initial synchronisation. I haven't found any circumstances yet where it changes from `synchronised` back to `unsynchronised`.

For example, I have a system configured with a 32s chrony polling period where I have added an `iptables` rule to block incoming traffic on UDP port 123. The system has been running for over 68 hours now without any NTP replies being received, yet `chrony waitsync` is still reporting that it is synchronised to the last valid time source.

Ensuring initial synchronisation at boot time should be straightforward, but I'm not sure how we would (or if we would want to) try and keep `chrony` and `timedatectl` permanently tracking each other with respect to synchronisation status.

markcorbinuk, 2020-11-16:
Looks like the latest release of `chrony` (4.0) may have some improvements in this area. From the `NEWS` file:
```
* Update clock synchronisation status and leap status more frequently
```
I'll build a new version for testing.

curcuz, 2020-12-16:
@markcorbinuk any news on this?

markcorbinuk, 2020-12-16:
@curcuz I've looked at the device from the support thread and can't see any reason why DBus/timedatectl should be reporting loss of sync. It looks `chrony` synchronised at boot (all sources gave 4 good replies) and it was only 50 minutes since the last received NTP packet when I looked. Usually it takes the kernel just under 9 hours to drop out of synchronisation if no replies are received. I will see if there are any other conditions that can cause the kernel to mark time as unsynchronised.

markcorbinuk, 2020-12-18:
I had a further look at the above support thread and found that `chrony` was reporting some strange values for `root dispersion` and `skew`:
```
Reference ID    : ADF921CF (unifi.it-risch.de)
Stratum         : 3
Ref time (UTC)  : Wed Dec 16 14:01:13 2020
System time     : 0.000000000 seconds slow of NTP time
Last offset     : -0.000006721 seconds
RMS offset      : 0.000006721 seconds
Frequency       : 0.000 ppm slow
Residual freq   : +5.205 ppm
Skew            : 1000000.000 ppm
Root delay      : 0.046598878 seconds
Root dispersion : 3849.691650391 seconds
Update interval : 1.1 seconds
Leap status     : Normal
```
The `root dispersion` is a measure of the time difference between the server (unifi.it-risch.de) and it's upstream time source. Having looked at the `chrony` calculations for `maxerror` a value of 3849 for `root dispersion` is more that enough to exceed the maximum error value and cause the kernel to mark the time as unsynchronised. Hopefully this is just a one-off problem related to a serious network issue or glitch with this particular NTP server.

It is possible that this is the kind scenario that would normally be resolved quickly by `chrony` using a variable polling period. However, with our fixed 4.5 hour polling period we would have to monitor the kernel `timex` structure and run a 'burst' when the `maxerror` exceeds the threshold for synchronisation.

alexgg, 2020-12-21:
@markcorbinuk when you say "the kernel to mark the time as unsynchronised", do you mean that a user space app calling `adjtimex` will get a time not synchronized error? I am thinking that this is the moment when an iburst should happen, instead of having to monitor the `timex` structure. Probably chrony has a way of indicating this state too.

floion, 2021-09-01:
@markcorbinuk this should now be fixed, right?

markcorbinuk, 2021-09-06:
@floion This isn't fixed yet. The kernel synchronisation status is independent of the `chrony` status and at the moment it's not entirely clear how to best relate one to the other.