balena-os/meta-balena #1902: change default address pool of balena-engine to avoid collisions

robertgzr, 2020-05-20:
at some point in the past we were only using the default bridge for user applications, the subnet allocated to that was changed to avoid collisions: https://github.com/balena-os/meta-balena/commit/c376ee6fc09023bee5062a12faabd95038ff4bff

since then we moved to create a new bridge network for the application, this is not covered by `--bip` or `--fixed-cidr`

balena-engine assigns a subnet from https://github.com/balena-os/balena-libnetwork/blob/19.03-balena/ipamutils/utils.go#L18-L20 to bridge networks.
we should add `--default-address-pools` to set a new default set of pools to avoid collisions with the outside network.

---

https://www.flowdock.com/app/rulemotion/r-resinos/threads/U0O_nxYZ_wgK0TgzXuzThXMbTue

szony, 2020-07-10:
A prospec, Mutualink would like to have more control over the IP ranges for bridge networks as well. Just adding this info for the record.

robertgzr, 2020-07-30:
@szony I'm trying to find the thread but we concluded this can be done via network manager

majorz, 2022-06-21:
More specifically we need to make `/lib/systemd/system/balena.service` accept `--default-address-pool` through configuration. For example one should be able to specify `base=10.10.0.0/16,size=24` in the config.json file of the host OS.

zoobot, 2022-11-07:
https://github.com/moby/moby/pull/29376
https://github.com/moby/moby/pull/36054

It looks like balena-engine does have this whole PR:
https://github.com/moby/moby/pull/36396/files

https://github.com/balena-os/balena-engine/blob/master/cmd/dockerd/config_unix.go#L28
https://github.com/balena-os/balena-engine/blob/master/daemon/config/config.go#L117

I don't have a /etc/docker/daemon.json on my bob though
https://straz.to/2021-09-08-docker-address-pools/

@majorz What is handling that instead of the daemon.json?