balena-os/meta-balena #1903: Setting NTP server source from NTP pool fails to set the burst

alexgg, 2020-05-20:
As expalined in https://jel.ly.fish/728a48e0-84fa-4209-9509-88f613cbafd2, when adding an NTP server from config.json the following happens:
```
 /usr/bin/chronyc add server $ntp minpoll 14 maxpoll 14 || true
 /usr/bin/chronyc burst 4/10 "$ntp" || true
```
The second line works when $ntp is an NTP source, but not when it is an NTP pool that will assign a random NTP source.

Removing the second line with:
```
/usr/bin/chronyc burst 4/10
```
Should match every source instead and provide the desired effect.

markcorbinuk, 2020-06-25:
You can actually use the 'iburst' option with the chronyc 'add server' command. This is not documented in the chronyc man page, but has been present in the source code since v1.25 in 2011.

I think that in this case it would be reasonable to drop the separate 'burst' command and just use:
```
/usr/bin/chronyc add server $ntp iburst minpoll 14 maxpoll 14
```
This would mean that servers added while chronyd is running will use the same initial burst technique as the defaults ones specified in /etc/chrony.conf

(Note that the chronyc documentation is also out-of-date with respect to other options that you can now specify as part of the 'add server' command).

xginn8, 2020-06-25:
I think the larger issue here is that it is safer to treat any FQDN as a `pool` by default, rather than a `server`, whereas IPs can be considered `servers` without issue. This is related to https://github.com/balena-os/meta-balena/pull/1927 as well.

markcorbinuk, 2020-06-25:
I'll take a look at the implications of treating everything as a ```pool``` rather than a ```server``` and run some tests.

markcorbinuk, 2020-06-29:
The current release of chrony (v3.5) doesn't support the ability to add a ```pool``` via chronyc (it looks like this feature is coming in v4.0). Therefore any additional addresses added by the user via config.json will have to be treated as a ```server``` for the time being.

Some basic testing has indicated that when/if we update to chrony v4.0 it should be possible to treat any FQDN or IP address as a ```pool``` without any detrimental effects.

closed 2020-06-30 by bulldozer-balena[bot]