balena-os/meta-balena #1908: enable zram by default

shaunmulligan, 2020-06-01:
enable zram following https://haydenjames.io/raspberry-pi-performance-add-zram-kernel-parameters/  and change sysctl to use swap more.

It definitely feels like it made things more useable and stable on the rpi0.

shaunmulligan, 2020-06-02:
From the arch call:
- We will enable this from all device types
- We should commit 25% to compressed, or 1GB which ever one is smallest, so that for device types with lots of ram we don't over commit to zram swap.

shaunmulligan, 2020-06-09:
@alexgg and @petrosagg in the blog post above they set:
```
vm.vfs_cache_pressure=500
vm.swappiness=100
vm.dirty_background_ratio=1
vm.dirty_ratio=50
```

Which seemed to work well, but I don't know enough about these settings, so do you think they make sense of should we set something more conservative? I guess they just force swap to be used sooner, so with zram enabled it shouldn't be a problem having them set as above?

alexgg, 2020-06-10:
hi @shaunmulligan I think this type of tweak are either device specific or application/load specific. My thinking was to leave the kernel defaults in meta-balena and leave to integration layers or even application containers tweaking these if required.

My intuition tells me that giving preference to compressed swap will make a system with enough free RAM slower, and it may only be needed in the cases where memory is scarce, in which case the default settings will also kick in. But empirical testing for specific applications may well benefit from some tweaking.

For reference, https://documentation.suse.com/sles/15-SP1/html/SLES-all/cha-tuning-memory.html#cha-tuning-memory-vm

shaunmulligan, 2020-06-11:
@alexgg yeah thats very fair and we can do the more aggressive settings on devices like the rpi0 if need be 👍

commit 46a2ea58 (2020-06-08): packagegroup-resin: Enable compressed swap

Enable compressed swap using zram by default.

Fixes #1908

Change-type: minor
Changelog-entry: Enable compressed swap
commit e9f6fea1 (2020-06-08): packagegroup-resin: Enable compressed swap

Enable compressed swap using zram by default.

Fixes #1908

Change-type: minor
Changelog-entry: Enable compressed swap
commit b8cd041d (2020-06-10): systemd: Add compressed memory swap support

This enables a compressed RAM swap of 25% of total system memory or 1GB,
whichever smallest.

Fixes #1908

Change-type: minor
Changelog-entry: Add compressed memory swap support
commit bbb1ab41 (2020-06-10): packagegroup-resin: Enable compressed swap

Enable compressed swap using zram by default.

Fixes #1908

Change-type: minor
Changelog-entry: Add compressed memory swap support
closed 2020-07-02 by bulldozer-balena[bot]