balena-os/meta-balena #1910: review balena-engine systemd service

robertgzr, 2020-06-02:
we currently diverge from the one provided by upstream here:
https://github.com/moby/moby/blob/master/contrib/init/systemd/docker.service

* `MountFlags=slave`
  preventing us from using live restore functionality
  moby/moby#22806 (comment)
  access.redhat.com/articles/2938171

* `LimitNOFILE=infinity`, `LimitNPROC=infinity`
  not-insignificant performance overhead due to
  limits being propagated to all children (containerd + containers)
  moby/moby@8db6109

* `Delegate=yes`
  allow docker to manage it's cgroup subtree without systemd
  interference
  moby/moby#20152
  moby/moby@d16737f

* `TasksMax=infinity`
  prevent systemd from setting a default task limit of 512 on the engine
  cgroup, on linux >=4.3
  systemd/systemd#1239
  systemd/systemd#1886

* `TimeoutStartSec=0`
   prevent systemd from interrupting slow engine startup, potentially rendering a        device unusable because we never get the supervisor back up
   moby/moby@eff5e6498fa424587c339b5ae54e28f003775e4b

as per my previous commit here: see: https://github.com/balena-os/meta-balena/commit/fb2b63e5da8d2d57701501ffe88067a86928842a

robertgzr, 2020-06-02:
regarding `TimeoutStartSec`

1. there seems to be an issue where some low-spec devices can exceed this timeout, which triggers systemd to stop the unit, see https://www.flowdock.com/app/rulemotion/balenalabs/threads/chG_M67A0Va0Bl2cgLiZ4pdCelh

2. upstream introduced this setting when they put their graphdriver migration code into the engine startup code. In the future we will also run similar migrations, see https://github.com/balena-os/balena-engine/pull/168. So disabling it seems like a good move.

robertgzr, 2020-06-04:
Are there any counter arguments to having live-restore enabled on device?

roman-mazur, 2020-07-16:
Just applied the `TimeoutStartSec=0` change to mitigate an issue on a RPi Zero device that could not get the engine running in 90 seconds (the app is balenaSense, so it takes time to load 5 containers).

alexgg, 2022-04-29:
This refactoring happened in https://github.com/balena-os/meta-balena/commit/8227a61f6bef6d93cc6a5acd0ef93a2012079964.

The `TimeoutStartSec` was disabled in https://github.com/balena-os/meta-balena/commit/c4ce541526455c8cb891e7c76889e0d4cb2735b1

closed 2022-04-29 by alexgg