balena-os/meta-balena #1915: UDP packets not being forwarded by redsocks

sradevski, 2020-06-03:
User reports that the corresponding iptables rules exist for TCP, but not for UDP.

See: https://app.frontapp.com/open/cnv_7izqlu5

alexgg, 2020-06-04:
@sradevski @majorz I tend to think this is by design. The documented purpose of the proxy functionality is to allow a Balena device to connect to BalenaCloud when it sits behind a proxy. Forwarding UDP packets (apart from DNS which is tunnelled through TCP) is not required for this. 
I will go back to the customer and ask for some background in the use case.

majorz, 2020-06-04:
This PR fixes proxy usage for containers: https://github.com/balena-os/meta-balena/pull/1056. To me the intuitive usage is to serve all traffic (opposed to how the VPN is setup). With the current setup TCP streams from containers will go through the proxy, but UDP packets won't, and this sounds like a bug.

levesquejf, 2020-06-04:
Would it be possible to have it configurable with a file in the `/mnt/boot/system-proxy/` directory?