balena-os/meta-balena #1918: Better issue reporting in update log as to why there was a rollback

etcetc, 2020-06-04:
I recently udpated a device from 2.46 to 2.48.  After reboot, I realized that the device remained at 2.46.  The log file in /mnt/data/balenahup seems to indicate that everything is OK until

```
[INFO] Deploying /mnt/boot/overlays/w1-gpio-pullup.dtbo... done.
[INFO] Deploying /mnt/boot/overlays/i2c-rtc-gpio.dtbo... done.
[INFO] Deploying /mnt/boot/splash/resin-logo.png...file blacklisted. Ignoring.
[INFO] Deploying /mnt/boot/config.txt...file blacklisted. Ignoring.
[INFO] Deploying /mnt/boot/start.elf... overwriting modified file... done.
[INFO] Running rollback hook 
[INFO] Switching uboot root partition index to 3...... done.
New/Forward hooks (new os container) ran successfully.
After hooks (old os) ran successfully
[000000554][LOG]Supervisor update start...
[000000554][LOG]No explicit supervisor version was provided, update to default version in target balenaOS...
[000000591][LOG]Extracted default version is v10.8.0...
[000000592][LOG]Supervisor update: will be upgrading from v10.6.27 to v10.8.0
[000000594][LOG]Extracted supervisor vars: ID: 5841
[000000594][LOG]Setting supervisor version in the API...
[000000595][LOG]Running supervisor updater...
[000000722][LOG]Stopping supervisor and related services...
[000000733][LOG]No update package cleanup done
[000000733][LOG]Rebooting into new OS in 5 seconds...
```

If the lines:
```
[INFO] Running rollback hook 
[INFO] Switching uboot root partition index to 3...... done.
New/Forward hooks (new os container) ran successfully.
After hooks (old os) ran successfully
```

are indicating that there was an issue and so OS rollback was initiated (which is not clear from the text), then there should be some indication as to what the problem was because in the default case the data is not persisted across boots.

floion, 2020-06-04:
Actually, small correction, the healthchecks are run after the device reboots in the new OS so `[INFO] Running rollback hook` is not an indication of this happening. We should be enhancing the healthcheck services to also save the reason for the failure on disk so it's available after it rollbacks.

alexgg, 2021-07-15:
The rollback-health script is already verbose enough - enabling persistent logging and looking at the last boot is the only thing needed, the cause for rollback is clearly stated in the logs.
Altboot rollback is more complicated because usually the OS won't start so we won't have logging journal. To debug this case  we are looking into using `pstore` and `ramoops` and a memory buffer. This work is being tracked in https://github.com/balena-os/meta-balena/blob/master/meta-balena-common/recipes-core/balena-rollback/files/rollback-health

closed 2021-07-15 by alexgg