balena-os/meta-balena #1922: Revisit question of  inactive partition cleanup: can we speed up hostOS updates the second time around?

saintaardvark, 2020-06-12:
Currently, we're using deltas to fetch host OS updates.  If we encounter problems (health rollbacks, for example),  attempting the OS update a second time sometimes means downloading the delta a second time.  It would be good if we could skip that and re-use (with proper verification) the already-downloaded delta; this would make for a quicker update and less bandwidth usage.

xginn8, 2020-06-26:
I have been experimenting with this locally for a few days and I think in most cases it doesn't make sense to leave around old images since space is so tight and we in large part are using layers from the active partition to seed the download via delta (in the case of HUP). The _only_ case where this really makes sense is in fact a subsequent pull of the same exact image/metadata. I don't know if the added complexity of comparing locally-available and new target hostapps is worth the small optimization in the repull case.

alexgg, 2021-07-15:
Closing this as per Matt's comment above.

closed 2021-07-15 by alexgg