balena-os/meta-balena #1977: Data partition filesystem resize fails for large disks

alexgg, 2020-08-05:
When provisioning on large disks the filesystem resize fails with:

```
Jul 28 14:17:44 localhost sh[740]: Resizing the filesystem on /dev/disk/by-state/resin-data to 1952801880 (1k) blocks.
Jul 28 14:17:44 localhost sh[740]: The filesystem on /dev/disk/by-state/resin-data is now 1952801880 (1k) blocks long.
Jul 28 14:17:44 localhost kernel: EXT4-fs (sda6): ext4_check_descriptors: Block bitmap for group 131056 not in group (block 35184616189952)!
Jul 28 14:17:44 localhost kernel: EXT4-fs (sda6): group descriptors corrupted!
Jul 28 14:17:44 localhost resin-partition-mounter[758]: mount: /run/tmp.gN6HRW0bOF: mount(2) system call failed: Structure needs cleaning.
Jul 28 14:17:44 localhost resin-partition-mounter[758]: umount: /run/tmp.gN6HRW0bOF: not mounted.
Jul 28 14:17:44 localhost resin-partition-mounter[758]: resin-data: UNEXPECTED INCONSISTENCY; RUN fsck MANUALLY.
Jul 28 14:17:44 localhost resin-partition-mounter[758]:         (i.e., without -a or -p options)
Jul 28 14:17:45 localhost resin-partition-mounter[758]: WARN: UUIDs not regenerated - retry on next boot
Jul 28 14:17:45 localhost resin-partition-mounter[758]: INFO: Mounting /dev/disk/by-state/resin-data (resin-data) in /mnt/data.
Jul 28 14:17:45 localhost kernel: EXT4-fs (sda6): ext4_check_descriptors: Block bitmap for group 131056 not in group (block 35184616189952)!
Jul 28 14:17:45 localhost kernel: EXT4-fs (sda6): group descriptors corrupted!
Jul 28 14:17:45 localhost resin-partition-mounter[758]: mount: /mnt/data: mount(2) system call failed: Structure needs cleaning.
```

This might be related to `resize2fs`, see https://github.com/rancher/k3os/issues/145

For details see https://jel.ly.fish/2399d089-df1b-4d01-b2f3-26ae4cda77af

floion, 2020-08-07:
Ordered 1 https://www.amazon.co.uk/Seagate-BarraCuda-Internal-Drive-Factor/dp/B01LXRWWB6/ref=sr_1_fkmr0_1?dchild=1&keywords=Seagate+2TB+BarraCuda+2.5%22+hdd+model+ST2000LM015+%28SATA+III%29&qid=1596786027&sr=8-1-fkmr0 and will investigate once we get the hw

alexgg, 2022-08-11:
Addressed in the linked PR, closing.

closed 2022-08-11 by alexgg