balena-os/meta-balena #1978: Update ModemManager and related libraries (linqmi, libmbim) to latest stable

alexgg, 2020-08-05:
This is a customer request for 1.14.0 or newer.

See https://www.flowdock.com/app/rulemotion/r-hardware/threads/5Ln_JrwpzfJf5dKF73YxU_cXel-

shaunmulligan, 2020-08-05:
MM 1.14.0 changelog: https://cgit.freedesktop.org/ModemManager/ModemManager/tree/NEWS

This one is the one that worries me:
> * The daemon switched to 'STRICT' filter mode by default. The old 'DEFAULT'
   mode is renamed to 'LEGACY' and is considered now deprecated. Under the
   'STRICT' filter mode, the TTY blacklist and greylist are ignored, and the
   port probing mechanism uses its own heuristics to guess whether a given TTY
   is owned by a modem or not.

alexgg, 2020-09-10:
This has been done for v2.57.0.
https://github.com/balena-os/meta-balena/commit/1c80d6cc69d687fab6f2385efca5d6564c28370d

closed 2020-09-10 by alexgg