balena-os/meta-balena #1979: Firmware crash on iwlwifi wireless interfaces

alexgg, 2020-08-05:
Reproduced on an Intel NUC with the following device:
```
iwlwifi 0000:3a:00.0: Detected Intel(R) Dual Band Wireless AC 8265, REV=0x230
```

Seems to be caused by high traffic, so a stress transfer test should trigger it.

```
[  +0.000013] ieee80211 phy0: Hardware restart was requested
[  +0.514791] iwlwifi 0000:3a:00.0: Applying debug destination EXTERNAL_DRAM
[  +0.133007] iwlwifi 0000:3a:00.0: Applying debug destination EXTERNAL_DRAM
[  +0.066939] iwlwifi 0000:3a:00.0: FW already configured (0) - re-configuring
[  +0.330487] iwlwifi 0000:3a:00.0: Microcode SW error detected.  Restarting 0x2000000.
[  +0.000150] iwlwifi 0000:3a:00.0: Start IWL Error Log Dump:
[  +0.000006] iwlwifi 0000:3a:00.0: Status: 0x00000080, count: 6
[  +0.000004] iwlwifi 0000:3a:00.0: Loaded firmware version: 36.77d01142.0
[  +0.000006] iwlwifi 0000:3a:00.0: 0x0000105C | ADVANCED_SYSASSERT          
[  +0.000004] iwlwifi 0000:3a:00.0: 0x000002E3 | trm_hw_status0
[  +0.000005] iwlwifi 0000:3a:00.0: 0x00000000 | trm_hw_status1
[  +0.000004] iwlwifi 0000:3a:00.0: 0x000248EC | branchlink2
[  +0.000004] iwlwifi 0000:3a:00.0: 0x0003A862 | interruptlink1
[  +0.000004] iwlwifi 0000:3a:00.0: 0x00000000 | interruptlink2
[  +0.000004] iwlwifi 0000:3a:00.0: 0xDEADBEEF | data1
[  +0.000004] iwlwifi 0000:3a:00.0: 0xDEADBEEF | data2
[  +0.000004] iwlwifi 0000:3a:00.0: 0xDEADBEEF | data3
[  +0.000004] iwlwifi 0000:3a:00.0: 0x00C0D8C7 | beacon time
[  +0.000004] iwlwifi 0000:3a:00.0: 0x8F4E558D | tsf low
[  +0.000004] iwlwifi 0000:3a:00.0: 0x00000006 | tsf hi
[  +0.000004] iwlwifi 0000:3a:00.0: 0x00000000 | time gp1
[  +0.000004] iwlwifi 0000:3a:00.0: 0x000563E9 | time gp2
[  +0.000004] iwlwifi 0000:3a:00.0: 0x00000001 | uCode revision type
[  +0.000004] iwlwifi 0000:3a:00.0: 0x00000024 | uCode version major
[  +0.000004] iwlwifi 0000:3a:00.0: 0x77D01142 | uCode version minor
[  +0.000004] iwlwifi 0000:3a:00.0: 0x00000230 | hw version
[  +0.000004] iwlwifi 0000:3a:00.0: 0x00C89000 | board version
[  +0.000004] iwlwifi 0000:3a:00.0: 0x0BE6001C | hcmd
[  +0.000004] iwlwifi 0000:3a:00.0: 0x02622002 | isr0
[  +0.000004] iwlwifi 0000:3a:00.0: 0x00000000 | isr1
[  +0.000004] iwlwifi 0000:3a:00.0: 0x08001802 | isr2
[  +0.000003] iwlwifi 0000:3a:00.0: 0x00400881 | isr3
[  +0.000004] iwlwifi 0000:3a:00.0: 0x00000000 | isr4
[  +0.000004] iwlwifi 0000:3a:00.0: 0x0BE5001C | last cmd Id
[  +0.000004] iwlwifi 0000:3a:00.0: 0x00000000 | wait_event
[  +0.000004] iwlwifi 0000:3a:00.0: 0x00000D45 | l2p_control
[  +0.000004] iwlwifi 0000:3a:00.0: 0x00009C20 | l2p_duration
[  +0.000004] iwlwifi 0000:3a:00.0: 0x00000003 | l2p_mhvalid
[  +0.000004] iwlwifi 0000:3a:00.0: 0x00000091 | l2p_addr_match
[  +0.000004] iwlwifi 0000:3a:00.0: 0x0000000D | lmpm_pmg_sel
[  +0.000004] iwlwifi 0000:3a:00.0: 0x19071249 | timestamp
[  +0.000004] iwlwifi 0000:3a:00.0: 0x00006068 | flow_handler
[  +0.000073] iwlwifi 0000:3a:00.0: Start IWL Error Log Dump:
[  +0.000005] iwlwifi 0000:3a:00.0: Status: 0x00000080, count: 7
[  +0.000004] iwlwifi 0000:3a:00.0: 0x00000070 | NMI_INTERRUPT_LMAC_FATAL
[  +0.000004] iwlwifi 0000:3a:00.0: 0x00000000 | umac branchlink1
[  +0.000004] iwlwifi 0000:3a:00.0: 0xC0086948 | umac branchlink2
[  +0.000004] iwlwifi 0000:3a:00.0: 0xC0083B0C | umac interruptlink1
[  +0.000004] iwlwifi 0000:3a:00.0: 0xC0083B0C | umac interruptlink2
[  +0.000004] iwlwifi 0000:3a:00.0: 0x00000800 | umac data1
[  +0.000004] iwlwifi 0000:3a:00.0: 0xC0083B0C | umac data2
[  +0.000004] iwlwifi 0000:3a:00.0: 0xDEADBEEF | umac data3
[  +0.000004] iwlwifi 0000:3a:00.0: 0x00000024 | umac major
[  +0.000003] iwlwifi 0000:3a:00.0: 0x77D01142 | umac minor
[  +0.000005] iwlwifi 0000:3a:00.0: 0xC088628C | frame pointer
[  +0.000003] iwlwifi 0000:3a:00.0: 0xC088628C | stack pointer
[  +0.000004] iwlwifi 0000:3a:00.0: 0x004E0118 | last host cmd
[  +0.000004] iwlwifi 0000:3a:00.0: 0x00000000 | isr status reg
[  +0.000028] iwlwifi 0000:3a:00.0: Fseq Registers:
[  +0.000017] iwlwifi 0000:3a:00.0: 0xACE0179F | FSEQ_ERROR_CODE
[  +0.000018] iwlwifi 0000:3a:00.0: 0x22F00D30 | FSEQ_TOP_INIT_VERSION
[  +0.000017] iwlwifi 0000:3a:00.0: 0x04C108A2 | FSEQ_CNVIO_INIT_VERSION
[  +0.000017] iwlwifi 0000:3a:00.0: 0x0000A10B | FSEQ_OTP_VERSION
[  +0.000018] iwlwifi 0000:3a:00.0: 0xEF473638 | FSEQ_TOP_CONTENT_VERSION
[  +0.000017] iwlwifi 0000:3a:00.0: 0xAC2FC300 | FSEQ_ALIVE_TOKEN
[  +0.000017] iwlwifi 0000:3a:00.0: 0xD2C36951 | FSEQ_CNVI_ID
[  +0.000017] iwlwifi 0000:3a:00.0: 0x3DE1FCE7 | FSEQ_CNVR_ID
[  +0.000018] iwlwifi 0000:3a:00.0: 0x00000010 | CNVI_AUX_MISC_CHIP
[  +0.000020] iwlwifi 0000:3a:00.0: 0x0BADCAFE | CNVR_AUX_MISC_CHIP
[  +0.000020] iwlwifi 0000:3a:00.0: 0x0BADCAFE | CNVR_SCU_SD_REGS_SD_REG_DIG_DCDC_VTRIM
[  +0.000020] iwlwifi 0000:3a:00.0: 0x0BADCAFE | CNVR_SCU_SD_REGS_SD_REG_ACTIVE_VDIG_MIRROR
[  +0.000018] iwlwifi 0000:3a:00.0: Collecting data: trigger 2 fired.
[  +0.000010] ieee80211 phy0: Hardware restart was requested
[  +0.509426] iwlwifi 0000:3a:00.0: Failing on timeout while stopping DMA channel 8 [0x07fd0003]
[  +0.019013] iwlwifi 0000:3a:00.0: Applying debug destination EXTERNAL_DRAM
[  +0.132382] iwlwifi 0000:3a:00.0: Applying debug destination EXTERNAL_DRAM
[  +0.067256] iwlwifi 0000:3a:00.0: FW already configured (0) - re-configuring
```

Similar to: https://bugs.archlinux.org/task/63967

Kernel patches to check:
* https://patchwork.kernel.org/patch/11029027/
* https://lore.kernel.org/linux-wireless/20191009094853.PfIm3J8o7DN_Femup-OXkJdtmKE7rftk1ODkm7cx-vk@z/

xginn8, 2020-08-05:
ref. https://www.flowdock.com/app/rulemotion/resin-starters/threads/EnKJD_-UK7-JC_yIF-DZqB-uuLs

xginn8, 2020-09-23:
@gelbal hit this too, ref. https://www.flowdock.com/app/rulemotion/resin-starters/threads/u2d1wvyTDGszF5Dc0g8DipFFCLF

jonathankehler, 2022-05-03:
https://bugzilla.kernel.org/show_bug.cgi?id=205061
Might it be related to this? We are seeing this with balena-os and an intel ax200