balena-os/meta-balena #1995: Shutdown and reboot not being handled correctly for systemd services

markcorbinuk, 2020-08-13:
The chronyd service is not being terminated correctly when a system shutdown or reboot are triggered.

When chrony is terminated it should update the 'last modified' time of the drift file so that system time can be restored from it at next boot. This update happens when you stop the service manually with `systemctl stop chronyd` but not when `shutdown` or `reboot` are used (or triggered via the dashboard).

When chrony is terminated manually the following is logged in the system journal:
`chronyd[1284]: 2020-08-13T13:24:25Z chronyd exiting`
However, this log message does not appear if you monitor the journal from a serial console while shutting down the system with shutdown or reboot.

Similarly, `dnsmasq` outputs the following log message when terminated manually:
`dnsmasq[1392]: dnsmasq[1392]: exiting on receipt of SIGTERM`
but this also does not appear when using shutdown or reboot.

The only service that has been observed to terminate as expected is journald:
`systemd-journald[808]: Received SIGTERM from PID 1 (systemd-shutdow).`

Tested on a RPI3 running balenaOS 2.52.7+rev1

markcorbinuk, 2020-08-17:
Looks like this might be a sequencing issue relating to unmounting of the filesystems.

I managed to capture some systemd debug logs for a `reboot` which show the following:
```[  177.478855] systemd[1]: Stopping Bind mount for /var/lib/chrony...```
...
```[  179.294847] systemd[1]: Unmounted /var/lib/chrony.```
...
```[  204.187663] systemd[1]: Stopping chronyd.service...```
...
```[  204.800463] systemd[1]: Stopped chronyd.service.```

It seems that '/var/lib/chrony' (which contains the drift file) is unmounted 25s before chrony is shutdown, meaning that the drift file is no longer available when chrony tries to update the file time.

alexgg, 2020-11-17:
@markcorbinuk is this issue resolved with your recent time related PRs?

markcorbinuk, 2020-11-17:
@alexgg Not yet. I'm looking at remote HTTPS header synchronisation next and then chrony issues after that.

closed 2022-01-17 by bulldozer-balena[bot]