balena-os/meta-balena #1999: Application containers get restarted after some time when RAM is being exhausted

floion, 2020-08-20:
When the device is low on memory (for example user is running memory consuming code in his app containers), the device gets in a state where it exhausted its available memory so the kernel is forced to continually toggle virtual memory between processes, hence a high kswapd usage, which in turns slows the system down significantly ultimately causing the applications to restart and the engine/supervisor will fail to successfully restart them:

systemd[1]: resin-supervisor.service: Watchdog timeout (limit 3min)!
systemd[1]: balena.service: Watchdog timeout (limit 6min)!
systemd[1]: Failed to start Balena Application Container Engine.
systemd[1]: Failed to start Balena supervisor.

User reported he induced the low memory state by running `stress-ng --vm 1 --vm-bytes 500M` in addition to his usual containers just to make it crash faster.

federico-d, 2021-10-31:
`______________________MEMORY:
MemTotal:         956244 kB
MemFree:           14012 kB
MemAvailable:      94020 kB
Buffers:            1992 kB
Cached:           130412 kB
SwapCached:            0 kB
Active:           740456 kB
Inactive:          61324 kB
Active(anon):     689704 kB
Inactive(anon):    29484 kB
Active(file):      50752 kB
Inactive(file):    31840 kB
Unevictable:           0 kB
Mlocked:               0 kB
SwapTotal:             0 kB
SwapFree:              0 kB
Dirty:               144 kB
Writeback:             0 kB
AnonPages:        669340 kB
Mapped:            66204 kB
Shmem:             50480 kB
Slab:              99260 kB
SReclaimable:      30084 kB
SUnreclaim:        69176 kB
KernelStack:        5348 kB
PageTables:         7892 kB
NFS_Unstable:          0 kB
Bounce:                0 kB
WritebackTmp:          0 kB
CommitLimit:      478120 kB
Committed_AS:    3327024 kB
VmallocTotal:   135290290112 kB
VmallocUsed:           0 kB
VmallocChunk:          0 kB
Percpu:             3288 kB
HardwareCorrupted:     0 kB
AnonHugePages:    276480 kB
ShmemHugePages:        0 kB
ShmemPmdMapped:        0 kB
CmaTotal:          24576 kB
CmaFree:             352 kB
HugePages_Total:       0
HugePages_Free:        0
HugePages_Rsvd:        0
HugePages_Surp:        0
Hugepagesize:       2048 kB
Hugetlb:               0 kB
______________________CPU:
Mem: 942564K used, 13680K free, 50480K shrd, 1992K buff, 130812K cached
CPU:   0% usr   4% sys   0% nic  95% idle   0% io   0% irq   0% sirq
Load average: 2.51 1.45 1.20 1/333 5332
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
  965     1 root     S    1473m 158%   0% /usr/bin/balenad --experimental --log-driver=journald -s overlay2 -H fd:// -H unix:///var/run/balena.sock -H unix:///var/run/balena-engine.sock --dns 10.114.102.1 --bip 10.114.101.1/24 --fixed-cidr=10.114.101.0/25 --max-download-attempts=10 --exec-opt native.cgroupdriver=systemd
  843   724 root     S    1407m 150%   0% node index.js
 1113  1097 root     S    1019m 109%   0% node dist/server.js
  995   965 root     S    1009m 108%   0% balena-engine-containerd --config /var/run/balena-engine/containerd/containerd.toml --log-level info
  896   737 root     S     974m 104%   0% node server.js
  657     1 root     S     937m 100%   0% balena-engine-containerd-shim -namespace moby -workdir /var/lib/docker/containerd/daemon/io.containerd.runtime.v1.linux/moby/e2c27a872ec31b5ad0b7345b446b464355a704a72585d7f28a6932314af69aef -address /var/run/balena-engine/containerd/balena-engine-containerd.sock -containerd-binary /usr/bin/balena-engine -containerd-binary-argv0 balena-engine-containerd -runtime-root /var/run/balena-engine/runtime-runc
  641     1 root     S     937m 100%   0% balena-engine-containerd-shim -namespace moby -workdir /var/lib/docker/containerd/daemon/io.containerd.runtime.v1.linux/moby/ec59ff2ce1c0f3b51ba93c05b09ba98e2514d20379dd7e3ac6dbe27c0609e177 -address /var/run/balena-engine/containerd/balena-engine-containerd.sock -containerd-binary /usr/bin/balena-engine -containerd-binary-argv0 balena-engine-containerd -runtime-root /var/run/balena-engine/runtime-runc
  642     1 root     S     937m 100%   0% balena-engine-containerd-shim -namespace moby -workdir /var/lib/docker/containerd/daemon/io.containerd.runtime.v1.linux/moby/264fac6c156ff3eab6e6b52228eebab34d65c24a9000318464ec63d4cb757a25 -address /var/run/balena-engine/containerd/balena-engine-containerd.sock -containerd-binary /usr/bin/balena-engine -containerd-binary-argv0 balena-engine-containerd -runtime-root /var/run/balena-engine/runtime-runc
 1629   995 root     S     865m  93%   0% balena-engine-containerd-shim -namespace moby -workdir /var/lib/docker/containerd/daemon/io.containerd.runtime.v1.linux/moby/0b35906c34224087713e0898dadcf73ef6f1cfb1370f43c5a6a06180dea1eb73 -address /var/run/balena-engine/containerd/balena-engine-containerd.sock -containerd-binary /usr/bin/balena-engine -containerd-binary-argv0 balena-engine-containerd -runtime-root /var/run/balena-engine/runtime-runc
 1611  1532 root     S     793m  85%   0% balena start --attach resin_supervisor
  837     1 root     S     791m  85%   0% balena-engine-containerd-shim -namespace moby -workdir /var/lib/docker/containerd/daemon/io.containerd.runtime.v1.linux/moby/7cfbf6706c842e52d0c7d8d296a9c111421758491496dbcae3c23c8369f7bf89 -address /var/run/balena-engine/containerd/balena-engine-containerd.sock -containerd-binary /usr/bin/balena-engine -containerd-binary-argv0 balena-engine-containerd -runtime-root /var/run/balena-engine/runtime-runc
14279     1 root     S     721m  77%   0% balena-engine-containerd-shim -namespace moby -workdir /var/lib/docker/containerd/daemon/io.containerd.runtime.v1.linux/moby/319d077d07ed07fcfca708b2521412ecac65f66ae44f41933dc08e6c70983e92 -address /var/run/balena-engine/containerd/balena-engine-containerd.sock -containerd-binary /usr/bin/balena-engine -containerd-binary-argv0 balena-engine-containerd -runtime-root /var/run/balena-engine/runtime-runc
  851   717 root     S     700m  75%   0% {node} npm
 1097  1096 root     S     535m  57%   0% node /usr/src/app/node_modules/.bin/cross-env NODE_ENV=production node dist/server.js
  256     1 root     S     385m  41%   0% /usr/sbin/NetworkManager --no-daemon
  181     1 root     S     307m  33%   0% /usr/sbin/ModemManager --log-journal
 1646  1629 root     S     281m  30%   0% node /usr/src/app/dist/app.js
[... 124 lines cut ...]
[... cut ...]
Oct 31 15:05:04 66d4a4c balenad[14421]:         /builds/1DyvR8Cc/0/logbot/iot/build_environments/edgedevices/raspberrypi3-64/balena-raspberrypi/build/tmp/work/aarch64-poky-linux/balena/19.03.13-dev+git074a481789174b4b6fd2d706086e8ffceb72e924-r0/git/src/import/.gopath/src/github.com/docker/docker/vendor/github.com/spf13/cobra/command.go:852 +0x228
Oct 31 15:05:04 66d4a4c balenad[14421]: github.com/docker/docker/vendor/github.com/spf13/cobra.(*Command).Execute(...)
...
______________________UPTIME:
 16:02:55  up 2 days  5:49,  0 users,  load average: 2.39, 1.44, 1.20`

klutchell, 2022-01-12:
Ongoing improvement is here (internal access only) https://jel.ly.fish/improvement-assign-higher-resource-priority-to-critical-os-tasks-8df3381f-0e02-4ac0-ac1a-ece04d125591