balena-os/meta-balena #2009: Could not generate persistent MAC address warnings confuse users that something is going wrong

sradevski, 2020-09-11:
User saw these messages being logged and they wondered if there is an issue with the device. Alex suggested that these are likely harmless, but it is still worth investigating if there is anything we can do about it, even if it is silencing the warnings.

klutchell, 2022-01-12:
We are currently shipping supervisor 244 and this warning should no longer appear.

closed 2022-01-12 by klutchell

thgreasi, 2022-01-12:
@klutchell can you clarify what `supervisor 244` is exactly?

dfunckt, 2022-01-12:
@klutchell can you please point me to the fix? I'd like to understand better before getting back to a user. What is "supervisor 244"?

klutchell, 2022-01-12:
Sorry, I meant systemd v244. Fixed typo above.

thgreasi, 2022-01-12:
@klutchell which balenaOS version was that bump in?