balena-os/meta-balena #2010: make sure the balena-healthcheck image exists

robertgzr, 2020-09-11:
we should make sure `balena-healthcheck-image` exists before failing the engine health check

robertgzr, 2020-09-11:
@acostach do you think the move to the separate service in #1961 could play a role in this?

dequis, 2021-01-25:
I just ran into this and reverse engineered my way into the solution (re-running `/usr/lib/balena/balena-healthcheck-image-load` manually).

Makes sense that moving it from `ExecStartPost` to an `oneshot` service would make this worse - before the problem would fix itself as soon as the balena systemd service restarts, but now it needs manual intervention.

FWIW, the "root cause" for the issue in my case is someone running "balena system prune -a" which deletes unused images.

I think it wouldn't hurt to include a call to /usr/lib/balena/balena-healthcheck-image-load in the healthcheck script? It doesn't do the "loading layer" thing if it exists:

```
# balena load -i /usr/lib/balena/balena-healthcheck-image.tar
2536d8d4e4b1: Loading layer [==================================================>]  6.656kB/6.656kB
# balena load -i /usr/lib/balena/balena-healthcheck-image.tar
Loaded image: balena-healthcheck-image:latest
# /usr/lib/balena/balena-healthcheck-image-load
Loaded image: balena-healthcheck-image:latest
# /usr/lib/balena/balena-healthcheck-image-load
Loaded image: balena-healthcheck-image:latest
# time /usr/lib/balena/balena-healthcheck-image-load
Loaded image: balena-healthcheck-image:latest

real    0m0.630s
user    0m0.394s
sys     0m0.075s
```

closed 2021-02-26 by klutchell